milkv-duo/duo-buildroot-sdk #2: What is the SSH password?

overheat, 2023-08-14:
I compiled the buildroot by myself.
Look the SSH username/password is not root/milkv.

weilinfox, 2023-08-15:
I tried to compiled the buildroot and boot it, the SSH  username/password is root/milkv.

frett27, 2023-08-19:
i got the same issue, using an ubuntu 22 as a desktop linux to connect to the milkv_duo board. 
ssh daemon respond to the query, but the milkv password for the root user don't work.
Are there restrictions on some cypher ?

frett27, 2023-08-19:
solved -> i was using the ip addr ip, whereas you have to use the ".1" as below 

`[185409.861639] rndis_host 3-2:1.0 enxc2866ec578c2: renamed from usb0
use@alexa:~/projets/2023-08-19_milkv_duo$ ip addr show enxc2866ec578c2
11: enxc2866ec578c2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether c2:86:6e:c5:78:c2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.96/24 brd 192.168.42.255 scope global dynamic noprefixroute enxc2866ec578c2
       valid_lft 2884sec preferred_lft 2884sec
    inet6 fe80::bb92:df34:6e95:7b0d/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
use@alexa:~/projets/2023-08-19_milkv_duo$ ssh 192.168.42.96
use@192.168.42.96's password: 

use@alexa:~/projets/2023-08-19_milkv_duo$ ssh root@192.168.42.96
root@192.168.42.96's password: 
Permission denied, please try again.
root@192.168.42.96's password: 

use@alexa:~/projets/2023-08-19_milkv_duo$ ssh root@192.168.42.1
The authenticity of host '192.168.42.1 (192.168.42.1)' can't be established.
ED25519 key fingerprint is SHA256:wlYM0yhjjsG2+PM5ptDcfj1zYgM8JS2GB2Ult3/EpmE.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.42.1' (ED25519) to the list of known hosts.
root@192.168.42.1's password: 
[root@milkv]~# 
`

frett27, 2023-08-21:
@overheat does it fix your issue ?

overheat, 2023-09-04:
yes.
Thanks!

closed 2023-09-04 by overheat

Sfinx, 2023-09-14:
Having the same problem with 30acdbc595f62 from Sep 11:

```
ifconfig enxd290b4303df4 192.168.42.2/24
ssh root@192.168.42.2
root@192.168.42.2's password: 
Permission denied, please try again.
...
ssh root@192.168.42.1
root@192.168.42.1's password: 
Permission denied, please try again.
```

overheat, 2023-09-14:
serial port for debugging is a good idea. Or just try to reboot it.

Sfinx, 2023-09-16:
Found the issue: had 192.168.0.x network configured at another interface. All is working !