milkv-duo/duo-buildroot-sdk #3: `undefined reference to `__atomic_compare_exchange_1'` when running `./build_milkv.sh`

VitalyAnkh, 2023-08-16:
After running `./build_milkv.sh`, I got error like this:
```
[riscv64-unknown-linux-musl-gcc] sample_venc.o
/home/vitalyr/projects/dev/embed/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/vitalyr/projects/dev/embed/duo-buildroot-sdk/middleware/v2/lib/libsys.a(cvi_vb.o): in function `CVI_VB_Init':
/root/.jenkins/workspace/v4.1.0_release_build/middleware/v2/modules/sys/src/cvi_vb.c:201: undefined reference to `__atomic_compare_exchange_1'
/home/vitalyr/projects/dev/embed/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/vitalyr/projects/dev/embed/duo-buildroot-sdk/middleware/v2/lib/libsys.a(cvi_vb.o): in function `CVI_VB_Exit':
/root/.jenkins/workspace/v4.1.0_release_build/middleware/v2/modules/sys/src/cvi_vb.c:225: undefined reference to `__atomic_compare_exchange_1'
collect2: error: ld returned 1 exit status
```
It seems to be already fixed by gcc toolchains([1], [2], [3]), but duo-build root-SDK will download an older gcc toolchain (gcc 10) to build the image. How about updating the toolchain?

1. https://github.com/riscv-collab/riscv-gcc/issues/12
2. https://www.overleaf.com/project/624c33c4e2d49b02be626e13
3. https://github.com/advancedtelematic/aktualizr/issues/1427#issuecomment-1128381452

pavelanni, 2023-09-01:
Same here. Exactly the same message as @VitalyAnkh posted.

Does anybody see a short-term solution?

carbonfix, 2023-11-22:
Try using the latest code. If the compilation still fails, it is recommended that you use Docker to compile:
https://github.com/milkv-duo/duo-buildroot-sdk#2-compiled-using-docker

closed 2023-11-22 by carbonfix

xnvi, 2024-06-24:
我自己的解决方法，在sophpi/middleware/v2/sample/venc/Makefile的EXTRA_LDFLAGS 后面添加 -latomic，对vdec中的makefile同样有效