milkv-duo/duo-buildroot-sdk #5: _call_kconfig_script:11: no such file or directory: /tmp/duo-buildroot-sdk/build/scripts/.py

kewer127, 2023-08-19:
I'm using default develop branch (commit: be2a69546e05700f669c758bf4540831555716ef), the script path is not correct. 

```bash
/tmp$ git clone https://github.com/milkv-duo/duo-buildroot-sdk.git
/tmp$ cd duo-buildroot-sdk
/tmp/duo-buildroot-sdk$ git log -1
commit be2a69546e05700f669c758bf4540831555716ef (HEAD -> develop, tag: Duo-V1.0.3, origin/develop, origin/HEAD)
Author: carbon <carbon@milkv.io>
Date:   Thu Aug 17 14:21:24 2023 +0800

    camera: support sensor gc2083
/tmp/duo-buildroot-sdk$ 
/tmp/duo-buildroot-sdk$ 
/tmp/duo-buildroot-sdk$ 
/tmp/duo-buildroot-sdk$ export MILKV_BOARD=milkv-duo
/tmp/duo-buildroot-sdk$ source milkv/boardconfig-milkv-duo.sh
/tmp/duo-buildroot-sdk$ source build/milkvsetup.sh
/proc/self/fd/14:8: cv1800b_board_sel: assignment to invalid subscript range
  -------------------------------------------------------------------------------------------------------
    Usage:
    (1) menuconfig - Use menu to configure your board.
        ex: $ menuconfig

    (2) defconfig $CHIP_ARCH - List EVB boards($BOARD) by CHIP_ARCH.
       ** cv183x ** -> ['cv1829', 'cv1832', 'cv1835', 'cv1838', 'cv9520', 'cv7581']
       ** cv182x ** -> ['cv1820', 'cv1821', 'cv1822', 'cv1823', 'cv1825', 'cv1826', 'cv7327', 'cv7357']
       ** cv181x ** -> ['cv181x', 'cv1823a', 'cv1821a', 'cv1820a', 'cv1811h', 'cv1811c', 'cv1810c', 'cv1810h', 'cv1812cp', 'cv1812h', 'cv1813h']
       ** cv180x ** -> ['cv180x', 'cv1800b', 'cv1800c', 'cv1801b', 'cv1801c', 'cv180zb']
        ex: $ defconfig cv183x

    (3) defconfig $BOARD - Choose EVB board settings.
        ex: $ defconfig cv1835_wevb_0002a
        ex: $ defconfig cv1826_wevb_0005a_spinand
        ex: $ defconfig cv181x_fpga_c906
  -------------------------------------------------------------------------------------------------------
/tmp/duo-buildroot-sdk$ menuconfig                 
 Run  function 
boards  common_functions.sh  cvisetup.sh  Kconfig  Makefile  milkvsetup.sh  output  panels  scripts  sensors  tools
_call_kconfig_script:11: no such file or directory: /tmp/duo-buildroot-sdk/build/scripts/.py
```

sidianneng, 2023-09-01:
It works fine with the latest one on ubuntu 22.04.

Rotule666, 2023-10-10:
Hello,

I am having the same problem

```
➜ duo-buildroot-sdk git:(develop) ✗ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy
```

```
duo-buildroot-sdk git:(develop) ✗ export MILKV_BOARD=milkv-duo
➜  duo-buildroot-sdk git:(develop) ✗ source milkv/boardconfig-milkv-duo.sh
➜  duo-buildroot-sdk git:(develop) ✗ source build/milkvsetup.sh
/proc/self/fd/19:8: cv1800b_board_sel: assignment to invalid subscript range
  -------------------------------------------------------------------------------------------------------
    Usage:
    (1) menuconfig - Use menu to configure your board.
        ex: $ menuconfig

    (2) defconfig $CHIP_ARCH - List EVB boards($BOARD) by CHIP_ARCH.
       ** cv183x ** -> ['cv1829', 'cv1832', 'cv1835', 'cv1838', 'cv9520', 'cv7581']
       ** cv182x ** -> ['cv1820', 'cv1821', 'cv1822', 'cv1823', 'cv1825', 'cv1826', 'cv7327', 'cv7357']
       ** cv181x ** -> ['cv181x', 'cv1823a', 'cv1821a', 'cv1820a', 'cv1811h', 'cv1811c', 'cv1810c', 'cv1810h', 'cv1812cp', 'cv1812h', 'cv1813h']
       ** cv180x ** -> ['cv180x', 'cv1800b', 'cv1800c', 'cv1801b', 'cv1801c', 'cv180zb']
        ex: $ defconfig cv183x

    (3) defconfig $BOARD - Choose EVB board settings.
        ex: $ defconfig cv1835_wevb_0002a
        ex: $ defconfig cv1826_wevb_0005a_spinand
        ex: $ defconfig cv181x_fpga_c906
  -------------------------------------------------------------------------------------------------------
```

The key error I think is that the script fails
```
/proc/self/fd/19:8: cv1800b_board_sel: assignment to invalid subscript range
```

when using menuconfig, the script name is missing but the path is good.
```
➜  duo-buildroot-sdk git:(develop) ✗ menuconfig
 Run  function
_call_kconfig_script:11: no such file or directory: /home/rotule/duo-buildroot-sdk/build/scripts/.py
```

I can use the  `build_milkv.sh` auto script successfully.

I am not very familiar with buildroot, i'm used to yocto, if you have a recommendation for lectures or steps to try, I would like to help solving this problem.

Kind regards,

GrootLiu, 2023-10-22:
我的版本是 2204 遇见了同样的问题
[image]
[image]

GrootLiu, 2023-10-22:
> 我的版本是 2204 遇见了同样的问题 [image] [image]

之后我尝试另一台机器，也出现了一样的问题
[image]

GrootLiu, 2023-10-22:
> > 我的版本是 2204 遇见了同样的问题 [image] [image]
> 
> 之后我尝试另一台机器，也出现了一样的问题 [image]

修改 duo-buildroot-sdk/build/common_functions.sh 文件能够解决这个问题，但是感觉治标不治本。

```sh
$ cd duo-buildroot-sdk
$ vim build/common_functions.sh

:%s.FUNCNAME\[0\]/0/g
:wq

$ source build/milkvsetup.sh
```

Rotule666, 2023-10-25:
building from a fresh container as proposer [here](https://github.com/milkv-duo/duo-buildroot-sdk/issues/3#issuecomment-1703022046) works as well

carbonfix, 2023-11-22:
Try using the latest code. If the compilation still fails, it is recommended that you use Docker to compile:
https://github.com/milkv-duo/duo-buildroot-sdk#2-compiled-using-docker

closed 2023-11-22 by carbonfix

hoyang, 2024-10-19:
use bash, not zsh, all ok now.