milkv-duo/duo-buildroot-sdk #8: serial bootloader scripts results in CRC errors

DAMEK86, 2023-09-06:
i try to use the serial boot ROM to speed up u-boot development.

however, i'm not able to download the `/fsbl/build/cv1800b_milkv_duo_sd/fip.bin` file by using `python3 cv181x_dl.py --serial --image_dir duo-buildroot-sdk/fsbl/build/cv1800b_milkv_duo_sd/fip.bin`

i modified the VID:PID parameters to match my USB2UART bridge controllers. however, i allways get CRC errors using the scripts

```
INFO: Using serial
INFO: CV181X USB download start

fip_path: duo-buildroot-sdk/fsbl/build/cv1800b_milkv_duo_sd/fip.bin/fip.bin
CV181X USB download start
Connecting to ROM
/dev/ttyUSB0USB port:  \    
USB VID:PID=10C4:EA60 SER=0001 LOCATION=1-1.3.2

done
Send cv_dl_magic.bin...
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 2e43
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 504c
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 5053
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 3120
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 6f61
ACK_CRC_ERROR cmd_crc 9b8; ret_crc d0a
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 616d
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 3a6c
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 3829
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 6172
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 2045
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 2d37
ACK_CRC_ERROR cmd_crc 9b8; ret_crc 2070
```

i tried this with the Silicon Labs CP210x UART Bridge (CR2102) found on an ESP32 as well as with an standalone FTDI FT232RL.

hope someone can help since the offical documentation (e.g. [CvitekBareandNon-BareChipBurningUpgradeOperationGuide_en.pdf](https://doc.sophgo.com/cvitek-develop-docs/master/docs_latest_release/CV180x_CV181x/en/01.software/OSDRV/Cvitek_Bare_and_Non-Bare_Chip_Burning_Upgrade_Operation_Guide/build/CvitekBareandNon-BareChipBurningUpgradeOperationGuide_en.pdf)) doesn't provide useful information about this topic..

thx

orangecms, 2024-04-26:
Hey, the protocol that the mask ROM offers appears on the serial that comea from the USB-C port.
UART0 only gets text output. What you are seeing there are ASCII characters.

To get into the USB loader mode, if you have a Duo and the IO board, isolate the USB pads that would attach to the USB hub on the IO board via the pogo pins, just using tape, and short the unpopulated pads next to the hole by the USB-C connector in the IO board - that would've been a second button, but they left it out, presumably because of the collision with the hub (my assumption).

I have written my own tool that lets you run a single flat binary on the primary core, based on the code supplied here, but simpler to use, which will help you develop a U-Boot SPL, if that is what you're after:
https://github.com/orangecms/sg_boot

I have added prebuilt binaries that just print to the UART for the Milk-V Duo (CV1800B), Duo 256M (SG2002) and Duo S (SG2000):
```
cargo run --release -- run oreboot_cv1800b.bin
```
or
```
cargo run --release -- run oreboot_sg200x.bin
```

DAMEK86, 2024-04-27:
thx for the information!

after i provided 3,3V to ADC2, and plugged in the USB type-C cable, my linux was able to detect the chip and your program run as expected

```
`target/release/sg_boot run oreboot_cv1800b.bin`
Payload size: 6160; aligned: 6656
Payload checksum: 94b8
Header checksum: c80a
Waiting for CVITek USB devices...
send NO MAGIC...
send HEADER...
Waiting for CVITek USB devices...
send NO MAGIC...
send PAYLOAD...
```

closed 2024-04-27 by DAMEK86

orangecms, 2024-04-27:
> thx for the information!
>
> after i provided 3,3V to ADC2, and plugged in the USB type-C cable, my linux was able to detect the chip and your program run as expected
>
> ```
> `target/release/sg_boot run [...]

Noice! With a USB serial converter attached to UART0, you should also see output on it. :) 

I just translated the first functions for DRAM init, setting up the PLLs and bringing up the controller. Are you working on upstream support for U-Boot SPL? I'd be happy to exchange, just let me know, and where - Matrix or some chat would be nice. I'm in the Milk-V Matrix channels, also on the RISC-V Slack.

DAMEK86, 2024-04-27:
yeah, the oreboot starts with
```
C.SCS/0/0.WD.URPL.USBI.USBW/1000.USBL.BS/EMMC.PS. E:load param1 (-78)
PS. E:load param1 (-78)
PS. E:load param1 (-78)
PS. E:load param1 (-78)
PS. E:load param1 (-78)
PS. E:load param1 (-78)
PS. E:load param1 (-78)
PS. E:load param1 (-78)
 E:Boot failed (8).
 E:ra=0x440a264
 E:RESET:panic:-1
WD.C.SCS/0/0.WD.URPL.USBI.USBK.USBD.BS/USB.PS.USBB.USBL.PE.BS.USBK.USBL.BE.J.B

oreboot 🦀 bt0
RISC-V arch 00000000
RISC-V core vendor: T-Head (0x05b7)
RISC-V implementation: unknown (0x00000000)
RISC-V hart ID 0
```

i like to build a image with yocto. i will start with latest u-boot now and see how far i come. didn't do anything with (c)oreboot so far.. 

I saw your post in the telegram channel already. but don't expect to much since i'm a bit busy currently.

again. thx a lot for streamlining the download tool!