milkv-duo/duo-buildroot-sdk #10: 编译img镜像，在ld链接时出现collect2: error: ld returned 1 exit status并蹦出，(CVI_VB_Exit)

guozijing-zjk, 2023-09-09:
# 环境信息

- OS: Endeavouros(Arch-Linux)
- SDK: 最新

# 复现步骤
> _由于是Arch,所以我尽量跟着sdk中需要的库和环境在Pacman和yay中进行安装。_

`#执行命令
./build_milkv.sh`

## 等待链接ld链接时出现：
[image]

## 好像是在link中middleware/v2/sample/venc的时候蹦出来的。
[image]
### 最后终端弹出collect2: error: ld returned 1 exit status，然后编译就终止蹦出来了，我该怎么办。:sob:，求指点是缺少什么库吗？
## 等待开发者的回信ing................

wxkj123, 2023-09-21:
Same problem as you.

fruit-lxe, 2023-10-13:
一模一样的问题，就是建立不了

hoyang, 2023-10-16:
试过这个码？[https://github.com/milkv-duo/duo-buildroot-sdk/issues/3#issuecomment-1703022046](https://github.com/milkv-duo/duo-buildroot-sdk/issues/3#issuecomment-1703022046)

carbonfix, 2023-11-22:
Try using the latest code. If the compilation still fails, it is recommended that you use Docker to compile:
https://github.com/milkv-duo/duo-buildroot-sdk#2-compiled-using-docker

closed 2023-11-22 by carbonfix

xnvi, 2024-06-24:
我自己的解决方法，在sophpi/middleware/v2/sample/venc/Makefile的EXTRA_LDFLAGS 后面添加 -latomic，对vdec中的makefile同样有效