milkv-duo/duo-buildroot-sdk #17: 在按照官方文档编译uboot时出现问题

asdvfsdf, 2023-10-10:
在编译官方uboot时执行build_uboot报错
https://doc.sophgo.com/cvitek-develop-docs/master/docs_latest_release/CV180x_CV181x/zh/01.software/OSDRV/U-boot_Porting_Development_Guide/build/html/3_U-boot_Transplant.html
`
duo-buildroot-sdk/u-boot-2021.10/arch/riscv/cpu/mtrap.S: Assembler messages:
duo-buildroot-sdk/u-boot-2021.10/arch/riscv/cpu/mtrap.S:65: Error: unrecognized opcode `csrr a0,scause', extension `zicsr' required
`
我以前遇到过这个错误，显示缺少zicsr模块，无法使特权指令，但是当我用自己的riscv64-unknown-elf-gcc和riscv64-unknown-linux-musl-gcc编译我自己写的一个简单的测试c文件时却显示没有问题，我的测试C文件如下
`
// test.c
int main(int argc, char *argv[])
{
  asm volatile ("csrr t0, misa");
  return 0;
}

`

asdvfsdf, 2023-10-10:
i am a stupid guy

carbonfix, 2023-10-24:
Have you found the cause of the issue?

closed 2023-11-22 by carbonfix