milkv-duo/duo-buildroot-sdk #20: makefile 9 all error 1

micropythondevel, 2023-10-17:
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:78: sample_venc] Error 1
make[1]: se sale del directorio '/home/icelon/risck/duo-buildroot-sdk/middleware/v2/sample/venc'
make: *** [Makefile:9: all] Error 1

carbonfix, 2023-10-24:
Please describe your compilation environment and the compilation steps.

Uro1, 2023-10-25:
I have had the same error with `duo-buildroot-sdk/middleware/v2/sample/venc`

Fresh build directory / clone of repo.

OS: Debian 12 Bookworm
```
$ uname -a
Linux Vanaheimr 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux
$ cmake --version  
cmake version 3.25.1
```

Error:
```
make[1]: Entering directory '/slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/sample/venc'
[riscv64-unknown-linux-musl-gcc] sample_venc.o
[riscv64-unknown-linux-musl-gcc] sample_venc_lib.o
[riscv64-unknown-linux-musl-gcc] sample_venc_testcase.o
[riscv64-unknown-linux-musl-gcc] sample_vdec_lib.o
[riscv64-unknown-linux-musl-gcc] sample_vdec_testcase.o
/slowstore/milkv/duo/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/lib/libsys.a(cvi_vb.o): in function `CVI_VB_Init':
/root/.jenkins/workspace/v4.1.0_release_build/middleware/v2/modules/sys/src/cvi_vb.c:201: undefined reference to `__atomic_compare_exchange_1'
/slowstore/milkv/duo/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/lib/libsys.a(cvi_vb.o): in function `CVI_VB_Exit':
/root/.jenkins/workspace/v4.1.0_release_build/middleware/v2/modules/sys/src/cvi_vb.c:225: undefined reference to `__atomic_compare_exchange_1'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:78: sample_venc] Error 1
make[1]: Leaving directory '/slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/sample/venc'
make: *** [Makefile:9: all] Error 1
Error: Build board milkv-duo failed!

```

I updated cmake to 3.27.7 in case that might have been the issue to rule it out:
```
$ cmake --version
cmake version 3.27.7
```
It resulted with the same error:
```
make[1]: Entering directory '/slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/sample/venc'
[riscv64-unknown-linux-musl-gcc] sample_venc.o
[riscv64-unknown-linux-musl-gcc] sample_venc_lib.o
[riscv64-unknown-linux-musl-gcc] sample_venc_testcase.o
[riscv64-unknown-linux-musl-gcc] sample_vdec_lib.o
[riscv64-unknown-linux-musl-gcc] sample_vdec_testcase.o
/slowstore/milkv/duo/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/lib/libsys.a(cvi_vb.o): in function `CVI_VB_Init':
/root/.jenkins/workspace/v4.1.0_release_build/middleware/v2/modules/sys/src/cvi_vb.c:201: undefined reference to `__atomic_compare_exchange_1'
/slowstore/milkv/duo/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/lib/libsys.a(cvi_vb.o): in function `CVI_VB_Exit':
/root/.jenkins/workspace/v4.1.0_release_build/middleware/v2/modules/sys/src/cvi_vb.c:225: undefined reference to `__atomic_compare_exchange_1'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:78: sample_venc] Error 1
make[1]: Leaving directory '/slowstore/milkv/duo/duo-buildroot-sdk/middleware/v2/sample/venc'
make: *** [Makefile:9: all] Error 1
Error: Build board milkv-duo failed!
```

closed 2023-11-06 by carbonfix