milkv-duo/duo-buildroot-sdk #24: Empty Partition

onkoe, 2023-10-29:
Hi! I just got my Duo, and I decided I wanted it to have my entire 120 GiB SD card. After poking through some files, I noticed that there's this `empty` partition, as shown in `milkv/genimage-milkv-duo.cfg`. 

Is there any purpose to that empty partition? I deleted it to expand the `rootfs`/"linux 0x83" partition, but I'm not sure if it's used for anything. 

Thanks for the great guide! :)

Uro1, 2023-11-01:
It is created in case you need swap, use that partition for it should you need it, it is explained in the `Getting Started` section of the documentation page - https://milkv.io/docs/duo/getting-started/swap

onkoe, 2023-11-03:
Great - thank you! Is it unreasonable to use [a swapfile](https://wiki.archlinux.org/title/swap#Swap_file) instead? 

That's what I've been doing for a long time on my devices, but if it's unrecommended, I'll ween off of it! 🤩️

Uro1, 2023-11-04:
The choice of swap file or partition depends on your own needs and use cases.

There's a ton of articles about swapfile vs swap partitions and the pro's & con's of each.
IMO I'd go for a swap partition over a swap file as:   
a) you would avoid any potential IO issues or amplifications from having to concurrently write to different areas of the same partition then when using a dedicated swap partition.  
b) MircoSD cards are cheap, even for larger cards 32GB++, creating a swap partition at the end of the drive can be done with gparted easily without denting much capacity, then you are free to resize the root partition to take up the rest of the drive, in the future you could then reasonably safely resize the swap partition should you need to as it is at the end of the drive.

closed 2023-11-22 by carbonfix