milkv-duo/duo-buildroot-sdk #30: Build Failure make: *** [scripts/fip_v2.mk:32: fsbl-build] Error 2

bebebib, 2023-11-12:
Env: WSL2 + Ubuntu22.04

Attempting to rebuild the kernel for the milkvduo, using WSL2 with Ubuntu22.04. I wanted to rebuild the kernel with gdbserver added so I set `CONFIG_TARGET_PACKAGE_GDBSERVER` to `y` in `build/boards/cv180x/cv1800b_milkv_duo_sd/cv1800b_milkv_duo_sd_defconfig`. But when attempting to build the kernel using the `build_milkv.sh` script, or the "step-by-step" way described [here](https://github.com/milkv-duo/duo-buildroot-sdk#step-by-step-compilation) I get the following error:

```
make[1]: *** [Makefile:196: duo-buildroot-sdk/fsbl/build/cv1800b_milkv_duo_sd/bl2/bl2.elf] Error 1
make[1]: Leaving directory 'duo-buildroot-sdk/fsbl'
make: *** [scripts/fip_v2.mk:32: fsbl-build] Error 2
Error: Build board milkv-duo failed!
```
I get this error regardless what I set `CONFIG_TARGET_PACKAGE_GDBSERVER` to, what could be the issue here? The error is referring to 
```
	${Q}$(MAKE) -j${NPROC} -C ${FSBL_PATH} O=${FSBL_OUTPUT_PATH} BLCP_2ND_PATH=${BLCP_2ND_PATH} \
		LOADER_2ND_PATH=${UBOOT_PATH}/${UBOOT_OUTPUT_FOLDER}/u-boot-raw.bin
```
So I cannot really tell what the issue is

bebebib, 2023-11-12:
Upon rerunning the build command with `sudo build_milkv.sh`, it seems successful, would there be any reason why I need `sudo` in the build command?

carbonfix, 2023-11-22:
Here are the things you need to pay attention to when compiling under WSL. You can try it to see if it can solve your problem:
https://github.com/milkv-duo/duo-buildroot-sdk#compiling-with-windows-linux-subsystem-wsl