milkv-duo/duo-buildroot-sdk #31: 不能使用ssh登录，提示密码不正确

RedLeaves699, 2023-11-15:
问题：
使用官方镜像v1.0.5
使用登录命令`ssh root@192.168.42.1`后输入密码milkv后会出现以下提示：

```bash
ssh root@192.168.42.1
root@192.168.42.1's password: 
Permission denied, please try again.
root@192.168.42.1's password: 
Permission denied, please try again.
root@192.168.42.1's password: 
root@192.168.42.1: Permission denied (publickey,password).
```
使用了boradIO扩展版连接网线

RedLeaves699, 2023-11-15:
本机操作系统信息
Linux debian01 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux

onkoe, 2023-11-19:
I fixed this issue for my custom buildroot by mounting the finished image, opening up `/etc/shadow`, making a new password hash with `openssl passwd -1 "milkv"`, and pasting the output hash between the first two colons after `root` in the file. 

On the other hand, you can also use serial to automatically login and change the password. That's [described here!](https://milkv.io/docs/duo/getting-started/setup#serial-console) ☺️

lztmfx, 2023-11-21:
我在windows10上也是无法通过ssh登录，IP为192.168.42.54，一直报错密码milkv不正确

RedLeaves699, 2023-11-28:
@onkoe  Thank you for your experience, it really helps.

I login milkv successfully now by serial and change the password of "root". Unfortunately, I still can't login by SSH on my Debian machine. But I login in "root" successfully with SSH on Windows Powershell, which makes me confused...

I checked the version of SSH on duo, it's `Dropbear v2020.81`, while the SSH version on my Debian machine is `OpenSSH_9.2p1 Debian-2+deb12u1, OpenSSL 3.0.11 19 Sep 2023`. Problems may arise here.

onkoe, 2023-11-28:
Glad it helped! Let’s see if we can get this fixed. 😄

In your `~.ssh/authorized_keys` file, is there an existing entry for the Duo’s IP? If so, try removing it! 

Otherwise, what seems to be the issue? Does it not connect at all, or is the password just not working?

RedLeaves699, 2023-11-28:
> Glad it helped! Let’s see if we can get this fixed. 😄
> 
> In your `~.ssh/authorized_keys` file, is there an existing entry for the Duo’s IP? If so, try removing it!
> 
> Otherwise, what seems to be the issue? Does it not connect at all, or is the password just not working?

The password just not working.

RedLeaves699, 2023-11-29:
I think I found some clues about the problem. I used `ssh -vvv root@192.168.42.1` on my Debian machine and my Windows10 machine, the difference of the result  is shown below:

Debian 12 (Can not login duo)
```bash
debug1: Local version string SSH-2.0-OpenSSH_9.2p1 Debian-2+deb12u1
debug1: Remote protocol version 2.0, remote software version OpenSSH_9.2p1 Debian-2+deb12u1
```

Windows10 (Successfully login duo)
```bash
debug1: Local version string SSH-2.0-OpenSSH_for_Windows_8.1
debug1: Remote protocol version 2.0, remote software version dropbear_2020.81
```

It seems that `OpenSSH_9.2p1 Debian-2+deb12u1` can't recognise the correct SSH version of milkv-duo.

closed 2024-01-14 by RedLeaves699