milkv-duo/duo-buildroot-sdk #44: no RNDIS on milkv duo 64MB Ram

protektwar, 2023-12-30:
Hello,

I have installed the img version 1.0.7 for my device, I can access the board over serial, I can also see the defaul ip adress, but I do not have any RNDIS connection under my Ubuntu dmesg.

Regards,
Alexandru

yevhenboichuk, 2023-12-30:
Did you try SSH?

protektwar, 2023-12-30:
I have tried and it is not working, as there is no rndis device in dmesg on my machine... I can see the default ip address over the serial interface.

drankinatty, 2024-01-06:
[Linux To Try Again To Disable All RNDIS Protocol Drivers](https://www.phoronix.com/news/Linux-Disable-RNDIS-Drivers) will be a continuing problem that the milkV developers will need to address. The likely cause for your problem is Ubuntu has blacklisted the rndis drivers to prevent them from loading. The once manually loaded, you then need to enable dhcp on the interface on the Ubuntu box so it receives an IP from the milkv-duo. Then you should be able to `ssh root@192.168.42.1` (or however you have yours configured)

You can check for blacklisted entries in your `/lib/modprob.d` directory (at least on openSUSE it is `/lib/modprobe.d/50-blacklist-rndis.conf`)

I suspect you are being hit by this issue. I was on openSUSE. Check to make sure you drivers are loaded, e.g.

```none
$ lsmod | grep rndis
rndis_wlan             69632  0
usb_f_rndis            49152  0
u_ether                32768  1 usb_f_rndis
libcomposite           81920  1 usb_f_rndis
rndis_host             20480  1 rndis_wlan
cdc_ether              24576  1 rndis_host
usbnet                 57344  3 rndis_wlan,rndis_host,cdc_ether
cfg80211             1040384  4 iwldvm,rndis_wlan,iwlwifi,mac80211
configfs               57344  3 usb_f_rndis,libcomposite
usbcore               327680  12 rndis_wlan,xhci_hcd,ehci_pci,usbnet,usbhid,cdc_acm,rndis_host,uvcvideo,ehci_hcd,btusb,xhci_pci,cdc_ether
```

(side-note, the buildroot-sdk builds fine on Archlinux as well as Ubuntu. The only stray package needed on Archlinux is `genext2fs` which is available in AUR)

protektwar, 2024-01-06:
I have checked for blacklist on /etc/modprob.d and /lib/modprob.d/. There is no indication of a blacklist on my machine. I'm able to load the rndis_host manually, still nothing...

[image]

black-ghost-off, 2024-01-08:
Try another image version

protektwar, 2024-01-08:
Up to now I have tried all the below images:
[image]

black-ghost-off, 2024-01-08:
The quality of USB?

drankinatty, 2024-01-08:
What is your device named? On Ubuntu, do `ip addr show` and look for the interface that was created. It may be `tty.USB0` or a normal interface name like `enp2384`. The key is that you need to tell Ubuntu that that device gets an IP via dhcp. On both openSUSE and Archlinux, the interface is created, but not enable (e.g. you will see the interface listed as `DOWN` in the `ip link show` output)

I don't know what config tool Ubuntu uses to configure network interfaces, but if it has one (`wicked`, `network-manager`, etc..), you should be able to configure that interface to get an IP via dhcp and then you will see your host gets an IP like `192.168.42.133`. Once you have an IP, then you can connect to the duo on `192.168.42.1`.

Side note: the Ethernet interface make things much easier. Your router (or whatever server you use on your LAN that hand out address) will just hand the duo and address on your local subnet. It helps to modify the `/etc/dhcpd.conf` on the duo so it provides `"milkv-duo"` as the hostname -- so your router will update its name resolution table and allow the duo to be reached via `ssh root@milkv-duo`. The changes needed are to uncomment `hostname` and comment `option host_name`, e.g. 

```none
...
# Inform the DHCP server of our hostname for DDNS.
hostname
...
# Request a hostname from the network
# option host_name
...
```

Post the output of `ip addr show` and `ip link show`.

protektwar, 2024-01-09:
As already specified, I can see the usb0 and the default ip address on the duo. Here is an printscreen taken from duo, over the UART connection:

[image]

I can't see anything on my Ubuntu machine related to RNDIS or related to any usb connection once I connect the duo to USB-C port on my machine. I have checked several cables nothing is happening.

drankinatty, 2024-01-09:
Your output on the duo (which is running Arch?) is fine. Ubuntu (your host) is the problem. If you see no interface on the Ubuntu box, and watching the journal when you plug the USB cable in doesn't do show anything other than a USB connection and no network interface setup -- then that is a problem. On the duo (Arch), the interface is set up, but is dnsmasq and dhcp running? (I've never used anything but the default duo image, so I don't know what the Arch image does)

Also remember, on the duo in `/mnt/system` there is a symlink for `usb.sh` that either is a link to `usb-rndis.sh` (meaning the duo USB will function as an RNDIS host with ethernet over USB) or it is a link to `usb-host.sh` (meaning RNDIS is disabled). If your link is to `usb-host.sh`, then that is why you computer isn't seeing the duo and not setting anything up when you plug it in. Look at:

```none
# ls -al /mnt/system/usb.sh
lrwxrwxrwx    1 root     root            12 Dec 23 03:37 /mnt/system/usb.sh -> usb-rndis.sh
```

That shows the proper link for RNDIS. if you see `/mnt/system/usb.sh -> usb-host.sh`, then you need to fix the link with `ls -sf  /mnt/system/usb-rndis.sh /mnt/system/usb.sh`,

Double check what you have.

protektwar, 2024-01-10:
It is as intended... 

[image]

protektwar, 2024-02-02:
I have bought a new board and used the same image as with the old one and I can connect over the RNDIS interface, I can see the usb0 ethernet interface on my Ubuntu machine...

closed 2024-02-02 by protektwar