milkv-duo/duo-buildroot-sdk #48: How do I access the kernel configuration in the build root

taspenwall, 2024-01-07:
I have the buildroot set up in a docker container and it works great to build the stock image.  I need to make some changes to the kernel and the rootfs.  I think I figured out how to use buildroot enough to add some app to the rootfs after loading a defconfig and running make menuconfig in the buildroot2021.05 folder.  What I can't get is how to how to configure the kernel.  I can go into the kernel folder and make a kernel but I have no idea what it's using as a config and where it's hidden.  If I had it I could then just cross compile the kernel in the linux-5.10 folder just like any kernel.  I'm glad milkv provided a buildroot but they hid everything under a thick layer of scripts and it makes it hard to change anything from the stock.

taspenwall, 2024-01-08:
I found the kernel config, and I know the output folder so I should be able to build a kernel, but editing the defconfig and running build_all script doesn't seem like the way it should be done.  I'm having lots of trouble with the buildroot to make the rootfs as I need to include one extra package.  When I run  `make milkv-duo-riscv64-musl-defconfig,` and then try to run `make` if fails to build as it's looking for a toolchain that is not present.  I guess this isn't how the scripts are making the rootfs.  I'd very much like to know the proper way to make changes to the rootfs and kernel.  I don't know where to turn for help so if anyone has any suggestions I'm all ears.

kylongmu, 2024-01-10:
Try this command: menuconfig_kernel

DaviVolcan, 2024-06-02:
Also, I am facing issues accessing the kernel menuconfig