milkv-duo/duo-buildroot-sdk #49: Build fail: make: *** [Makefile:584: br-rootfs-pack] Error 2

kylongmu, 2024-01-08:
Select with: 
3. milkv-duo256m
[image]

black-ghost-off, 2024-01-08:
https://github.com/milkv-duo/duo-buildroot-sdk?tab=readme-ov-file#compiling-with-windows-linux-subsystem-wsl

kylongmu, 2024-01-08:
> https://github.com/milkv-duo/duo-buildroot-sdk?tab=readme-ov-file#compiling-with-windows-linux-subsystem-wsl

According this guide, the build go more steps, but get new error:
 
[image]

taspenwall, 2024-01-08:
I get a similar failure when building the stock image for the duo in the docker containter.
 GEN     /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/Makefile
#
# configuration written to /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/.config
#
make[1]: Leaving directory '/home/work/buildroot-2021.05'
/home/work/buildroot-2021.05/utils/brmake -j4 -C /home/work/buildroot-2021.05
you need to install 'unbuffer' (from package expect or expect-dev)
make: *** [Makefile:584: br-rootfs-pack] Error 1

carbonfix, 2024-01-09:
@taspenwall This error requires pulling the latest Docker image

carbonfix, 2024-01-09:
@kylongmu Can you try to clone a completely new source code and try to compile it?

kylongmu, 2024-01-09:
> @kylongmu Can you try to clone a completely new source code and try to compile it?

I tried again, but it's still fail.
So I do this in Ubuntu 22.04 Host, build the repository in the NTFS disk, get the same error, but build success in the ext4 filesystem.
And I try again this with Docker, if the repository still in the host NTFS disk, it keeps fail, if I git clone in the Docker and build, it success.
So the result is that the fail just happen when the build under the NTFS filesystem.

carbonfix, 2024-01-10:
This SDK needs to be built in a Linux system, and the file system also needs to be a Linux file system (such as ext4). **The code must not be compiled in the NTFS file system!** If you are compiling in WSL, you also need to compile the code in the WSL default file system, not on the Windows NTFS file system such as the C drive or D drive mounted by WSL.

malud, 2024-01-12:
Hi, also facing this issue with MacOS (APFS (Encrypted)) + docker (buildx) with target `milkv-duo-lite`.

```
make[1]: Leaving directory '/home/work/buildroot-2021.05'
/home/work/buildroot-2021.05/utils/brmake -j5 -C /home/work/buildroot-2021.05
2024-01-12T19:02:35 >>> ncurses 6.1 Installing to target
Done in 1min 07s  (error code: 2)
make: *** [Makefile:584: br-rootfs-pack] Error 2
```

any ideas?

carbonfix, 2024-01-18:
@malud You can clone a copy of the code again to test whether the compilation is normal. If it is normal, it means that your old project needs to be cleaned and recompiled.

taspenwall, 2024-01-22:
I've pulled the latest docker image and I'm still having the same problem.  I know my repo and docker image are current as I just switch to a new workstation.

yaosheng79, 2024-02-27:
same issue here

platima, 2024-05-10:
Same issue here. WSL2, $PATH has no spaces, fresh pull of SDK, and build fails:

```make[1]: Leaving directory '/home/owner/duo-buildroot-sdk/buildroot-2021.05'
/home/owner/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j12 -C /home/owner/duo-buildroot-sdk/buildroot-2021.05
Done in 0s  (error code: 1)
make: *** [Makefile:605: br-rootfs-pack] Error 1```

melonedo, 2024-05-15:
> Same issue here. WSL2, $PATH has no spaces, fresh pull of SDK, and build fails:
> 
> ```
> /home/owner/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j12 -C /home/owner/duo-buildroot-sdk/buildroot-2021.05
> Done in 0s  (error code: 1)
> make: *** [Makefile:605: br-rootfs-pack] Error 1
> ```
> 
> Note, I did use `git apply --reject ../duo-lvgl-fb-demo/duo-kernel-fb_st7789v.patch`

It would be helpful to check build/br.log to determine the real cause.

Theskyisazureblue, 2024-07-22:
**Mayday, my Ubuntu20.04 file is in ext4 format, but I am having similar problems with the build**

# configuration written to /home/fmd/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/.config
#
make[1]: 离开目录“/home/fmd/duo-buildroot-sdk/buildroot-2021.05”
/home/fmd/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j4 -C /home/fmd/duo-buildroot-sdk/buildroot-2021.05
Done in 8s  (error code: 2)
make: *** [Makefile:597：br-rootfs-pack] 错误 2
Error: Build board milkv-duo-sd failed!

Theskyisazureblue, 2024-07-22:
Mayday, my Ubuntu20.04 file is in ext4 format, but I am having similar problems with the build

# configuration written to /home/fmd/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/.config
#
make[1]: 离开目录“/home/fmd/duo-buildroot-sdk/buildroot-2021.05”
/home/fmd/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j4 -C /home/fmd/duo-buildroot-sdk/buildroot-2021.05
Done in 8s  (error code: 2)
make: *** [Makefile:597：br-rootfs-pack] 错误 2
Error: Build board milkv-duo-sd failed!

melonedo, 2024-07-22:
> Mayday, my Ubuntu20.04 file is in ext4 format, but I am having similar problems with the build
> # configuration written to [...]

Check builld/br.log to look for any warning, this log message is uninformative.

zengxinqian, 2024-07-23:
when wsl has conda installed, i fix this issue by running 'conda install -c conda-forge Expect', ./build.sh milkv-duo-sd works, but ./build.sh milkv-duo256m-sd has another problem.

BoryaGames, 2024-07-25:
Here's my [br.log](https://mega.nz/file/obtllKLQ#1iihw4sG-_iw4aTSC3GeW4FfBHR2qzi5OmDxbzHQPQM), I have the same problem using WSL and using my VPS with Ubuntu 22.04 on ext4 too.
I found this part in the logs that maybe interesting:
```
2024-07-24T22:14:35 Making all in ntpd
2024-07-24T22:14:35 make[4]: Entering directory '/root/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duos-sd_musl_riscv64/build/ntp-4.2.8p15/ntpd'
2024-07-24T22:14:35 /usr/bin/make  all-am
2024-07-24T22:14:35 make[5]: Entering directory '/root/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duos-sd_musl_riscv64/build/ntp-4.2.8p15/ntpd'
2024-07-24T22:14:35 CC       cmd_args.o
2024-07-24T23:00:45 [01m[Kriscv64-unknown-linux-musl-gcc:[m[K [01;31m[Kfatal error: [m[KKilled signal terminated program cc1
2024-07-24T23:00:45 compilation terminated.
2024-07-24T23:00:45 Compile failed: command '/root/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duos-sd_musl_riscv64/per-package/python-lxml/host/bin/riscv64-unknown-linux-musl-gcc' failed with exit code 1
```

krzysztofkuczek, 2024-10-09:
same problem with du 256M sd.
/etc/wsl.conf setup corectly.
manual build during build_all

```
make[1]: Leaving directory '/home/qczek/milkv/duo-buildroot-sdk/buildroot-2021.05'
/home/qczek/milkv/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j14 -C /home/qczek/milkv/duo-buildroot-sdk/buildroot-2021.05
Done in 0s  (error code: 1)
make: *** [Makefile:597: br-rootfs-pack] Error 1
```