milkv-duo/duo-buildroot-sdk #54: pre-up ifconfig eth0 hw ether .... Not providing persistent MAC Address on IO Board

drankinatty, 2024-01-15:
I have followed the directions for providing a fixed MAC Address when using the IO Board modifying `/etc/network/interfaces` and adding:

```none
[02:00 milkv-duo:~] # cat /etc/network/interfaces
# interface file auto-generated by buildroot

auto lo
iface lo inet loopback

# added persistent hardware ethernet MAC
pre-up ifconfig eth0 hw 72:a2:bb:83:cf:12
```

And I've configured `dhcpd.conf` to send `hostname` so the server can update the local DNS with forward and reverse mapping for the IP address handed out to the duo. However, on reboot, the MAC address is different resulting in a different address being assigned each time. 

With `/etc/network/interfaces` as shown above, the local dhcp server sees:

```none
Jan 14 19:18:40 valkyrie dhcpd[810]: DHCPOFFER on 192.168.6.164 to 16:a7:65:c0:63:f3 (milkv-duo) via enp5s0
Jan 14 19:18:40 valkyrie dhcpd[810]: DHCPREQUEST for 192.168.6.164 (192.168.6.14) from 16:a7:65:c0:63:f3 (milkv-duo) via enp5s0
Jan 14 19:18:40 valkyrie dhcpd[810]: DHCPACK on 192.168.6.164 to 16:a7:65:c0:63:f3 (milkv-duo) via enp5s0
```

So despite having `pre-up ifconfig eth0 hw 72:a2:bb:83:cf:12` set in `/etc/network/interfaces`, the server sees the MAC Address of the duo as `16:a7:65:c0:63:f3`. 

Is there something else required to have the IO Board appear at a fixed MAC?

UncleGrumpy, 2024-01-15:
Did you try using the device `enp5s0` instead of `eth0` in `/etc/network/interfaces`? That would be the first thing I would try since that is the device being assigned the random MAC address.

MarkMLl, 2024-04-11:
> Did you try using the device `enp5s0` instead of `eth0` in `/etc/network/interfaces`? That would be the first thing I would try since that is the device being assigned the random MAC address.

enp5s0 is the device name being used on the host computer, not on the Duo.

It's working OK here, but note that the correct syntax is "...hw ether...", not just "hw" by itself. Was that a typo on your part (OP), or is there a documentation error somewhere?

orangecms, 2024-05-11:
similar: #27

drankinatty, 2024-05-12:
This can be closed - thank you for looking at it. It was a fat-finger error in the config file (my fault). Great little board.

closed 2024-05-12 by drankinatty