milkv-duo/duo-buildroot-sdk #58: GPIO26 and 27 has ADC, but does work?

pauloantuns, 2024-01-18:
The only "analogRead" function that i found on wiringx.h it's commented. (// int analogRead ... ) in duo-sdk/rootfs/usr/include/wiringx.h

There's any method to read analog data with the currently firmware version?
From the Duo Datasheet we can see the ADC pins.

[image]

MarkMLl, 2024-04-05:
Is GPIO27 available? According to https://github.com/milkv-duo/accessories/blob/master/Duo_USB%26Ethernet_IOB/duo_iob_v1.11.pdf (sheet 2) the IOB puts a pullup on it... I think the intention is that a pushbutton can optionally enable a network interface on USB even if the default is Ethernet.

pauloantuns, 2024-04-05:
Hi @MarkMLl !
Yes, GPIO 27 is available. I was able to find a solution to use the ADC module on the board and subimitted to merge. Check the github examples page:
https://github.com/milkv-duo/duo-examples/tree/main/adc

https://community.milkv.io/t/solved-does-duo-have-an-analog-read-function-from-the-gpio/1207