milkv-duo/duo-buildroot-sdk #60: What exactly is meant by "copy all to a blank tf card.." in how_to_download.txt, which is generated by the spinand configuration?

map588, 2024-01-24:
There is about 35M in the directory where "how_to_download.txt" is found, but the total project is like 10G.  There are lots of files compiled in build, in install, what is all??   I know a lot of it is redundant configurations/drivers for other devices, but I do not understand what or how to install.  

The physical positioning of the optional NAND flash footprint is right in front of the microSD card reader is, and I physically cannot put an SD card in this device, there is a SMD flash block in the way, and I would need to snap the card to complete the "how_to_download.txt" instructions.

carbonfix, 2024-01-24:
"Footprint in front of the microSD card" is not for installing spinor or spinand Flash. This is a reserved SD NAND pin. However, there is currently no way to burn firmware for it, so **please do not use it**.

yue-xiaomin, 2024-01-26:
@map588  SPINOR or SPINAND require a IOB, with pins connected to the SPI, specifically GPIOA22~27

map588, 2024-01-27:
Okay, I soldered the spiand onto the IO board, am I meant to literally copy the contents of only that one file onto an sd card?

yue-xiaomin, 2024-01-27:
IOB board SPI MOSI pin requires a pull-down resistor to ground, ranging from 4.7-10k, with a default setting of high and SPINOR startup mode. 

Yes, if you copy FIP and *. spinor or *. spinand to the memory card, the image will be automatically written to flash.

map588, 2024-01-27:
There are several spinand files in ./install after build, and there are copies of those files in update.zip.  Which set should go onto the sd card?  Should I initially boot with all files found in "install", and then after the initial install, copy over what is within update.zip?  I haven't found any documentation on how to install this way.

yue-xiaomin, 2024-01-27:
The *. spinand file for installation is packaged under update.zip, and the output after one click compilation is also output. You only need to copy the *. spinand file from the update compressed file

map588, 2024-01-27:
Okay!

map588, 2024-02-02:
> IOB board SPI MOSI pin requires a pull-down resistor to ground, ranging from 4.7-10k, with a default setting of high and SPINOR startup mode.
> 
> Yes, if you copy FIP and *. spinor or *. spinand to the memory card, the image will be automatically written to flash.

Are the pull down resistors meant to be where the two surface mount pads are which are empty near the NAND location?
[image]

yue-xiaomin, 2024-02-02:
[image]

map588, 2024-02-02:
@yue-xiaomin 
So if I wanted to use a NAND chip, I would remove R793 and solder it to R792?

yue-xiaomin, 2024-02-02:
> @yue-xiaomin 
> 
> So if I wanted to use a NAND chip, I would remove R793 and solder it to R792?

yes，Reserve R793 for spinor startup.

UncleGrumpy, 2024-02-02:
On future revisions of the IO Board it would be **so nice** if there were a dip switch to select `spinor` or `spinand`!

map588, 2024-02-03:
Is there a more comprehensive set of guides outside of this repo, and the project website?  It seems difficult to track relevant information down, such as the "menuconfig" option to set a more custom configuration for the build root.  I run into errors with u-boot often, for example.  The options are abbreviated, and some options either are incompatible configurations, when compiled in this way.
I have yet to compile the arm54 kernel for example.

yue-xiaomin, 2024-02-04:
> Is there a more comprehensive set of guides outside of this repo, and the project website? It seems difficult to track relevant information down, such as the "menuconfig" option to set a more custom [...]

Mainly refer to the configuration of other SDK boards and the documentation of chip manufacturers.
https://developer.sophgo.com/thread/471.html

map588, 2024-02-14:
There are a lot of blank solder pads on the I/O board, is there any extended functionality I can get by buying and soldering on more components to the board?