milkv-duo/duo-buildroot-sdk #61: Can not boot from nand spi

thiencuong, 2024-01-24:
[boot.log](https://github.com/milkv-duo/duo-buildroot-sdk/files/14040857/boot.log)
Hi.
The device can not boot from nand spi. i was attach the log file. please check.
Thank you.

yue-xiaomin, 2024-01-26:
IOB board SPI_ The MOSI pin requires a pull-down resistor to ground, ranging from 4.7-10k, with a default setting of high and SPINOR startup mode.

yue-xiaomin, 2024-01-26:
## Resetting to default environment
Start SD downloading...
switch to partitions #0, OK
mmc0 is current device
451584 bytes read in 22 ms (19.6 MiB/s)
addr 0000000082473000
fip.bin
do_cvi_update_spinand with version 0x1823a001
Scan and erase first 20 blocks
Generated checknum=0x3338e041
fip size=451584 bytes
write 1 copy of fip
write 2 copy of fip
Saving Environment to NAND... Erasing NAND...
Erasing at 0x880000 -- 100% complete.
Writing to NAND... OK
OK
64 bytes read in 2 ms (31.3 KiB/s)
Header Version:1

NAND erase.part: device 0 offset 0x280000, size 0x600000
Erasing at 0x860000 -- 100% complete.
OK
3371920 bytes read in 152 ms (21.2 MiB/s)

NAND write: device 0 offset 0x280000, size 0x337350
 3371856 bytes written: OK
nand nand speed 4.52 MB/s
64 bytes read in 2 ms (31.3 KiB/s)
Header Version:1

NAND erase.part: device 0 offset 0x8c0000, size 0x2800000
Erasing at 0x30a0000 -- 100% complete.
OK
14942272 bytes read in 664 ms (21.5 MiB/s)

NAND write: device 0 offset 0x8c0000, size 0xe40000
 14942208 bytes written: OK
nand nand speed 4.103 MB/s
64 bytes read in 3 ms (20.5 KiB/s)
Header Version:1

NAND erase.part: device 0 offset 0x30c0000, size 0x4600000
Erasing at 0x76a0000 -- 100% complete.
OK
15597632 bytes read in 694 ms (21.4 MiB/s)

NAND write: device 0 offset 0x30c0000, size 0xee0000
 15597568 bytes written: OK
nand nand speed 4.30 MB/s
64 bytes read in 2 ms (31.3 KiB/s)
Header Version:1

NAND erase.part: device 0 offset 0x76c0000, size 0x400000
Erasing at 0x7aa0000 -- 100% complete.
OK
2097216 bytes read in 96 ms (20.8 MiB/s)

NAND write: device 0 offset 0x76c0000, size 0x200000
 2097152 bytes written: OK
nand nand speed 4.112 MB/s

This is my update process.

thiencuong, 2024-01-26:
Thank you.
I using the NAND FLASH (W25N01GVxxIG/IT)
I meet this error when booting from NAND

WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46
WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46
WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46
WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46
WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46
WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46
WD.C.SCS/0/0.WD.URPL.USBI. E:TRAP/0xce01.
 E:RESET:plat/phobos/platform.c:46

thiencuong, 2024-01-27:
It is booted from nand now. thank you

closed 2024-01-29 by thiencuong