milkv-duo/duo-buildroot-sdk #62: make: *** [Makefile:604: br-rootfs-pack] Error 2 Error: Build board milkv-duo failed!

muditmadan, 2024-01-25:
I keep getting this error when I try to compile using wsl 2

make: *** [Makefile:604: br-rootfs-pack] Error 2
Error: Build board milkv-duo failed!

carbonfix, 2024-01-26:
You can clone a copy of the code again to test whether the compilation is normal. If it is normal, it means that your old project needs to be cleaned and recompiled.

xpww, 2024-02-01:
I also have the same problem, have you solved it?

xpww, 2024-02-01:
问题已经解决，是因为conda环境的问题，可以conda install conda-forge::expect或者conda deactivate后再执行脚本

jiapei100, 2024-02-07:
> 问题已经解决，是因为conda环境的问题，可以conda install conda-forge::expect或者conda deactivate后再执行脚本

Nope...  我没有安装 conda ....

thirstyone, 2024-02-13:
same error. here's the end of br.log: 
```
2024-02-13T16:32:09 patching file test/aclocal.m4
2024-02-13T16:32:09 patching file test/configure
2024-02-13T16:32:09 patching file test/tclock.c
2024-02-13T16:32:09 
2024-02-13T16:32:09 Applying 0001-gcc-5.x-MKlib_gen.patch using patch:
2024-02-13T16:32:09 patching file ncurses/base/MKlib_gen.sh
2024-02-13T16:32:09 Hunk #1 succeeded at 501 (offset -4 lines).
2024-02-13T16:32:09 >>> host-ncurses 6.1 Updating config.sub and config.guess
2024-02-13T16:32:09 for file in config.guess config.sub; do for i in $(find /duo-buildroot-sdk-Duo-V1.0.8/buildroot-2021.05/output/milkv-duo256m-lite_musl_riscv64/build/host-ncurses-6.1 -name $file); do cp support/gnuconfig/$file $i; done; done
2024-02-13T16:32:10 >>> host-ncurses 6.1 Patching libtool
2024-02-13T16:32:10 make[1]: Leaving directory '/duo-buildroot-sdk-Duo-V1.0.8/buildroot-2021.05'

```

Plaque-fcc, 2024-02-13:
> You can clone a copy of the code again to test whether the compilation is normal. If it is normal, it means that your old project needs to be cleaned and recompiled.

Yeah, cloning (and downloading everything again) fixes the problem.

For me, this happened tonight, and last time I build a firmware was v1.0.7 somewhere.

Freireg, 2024-02-28:
I'm having the same issue with V1.0.8 and V1.0.9 while attempting to build from a WSL ubuntu 22. And if I try to build directly from buildroot I receive the following: 
```
➜  duo-buildroot-sdk git:(Duo-V1.0.8) defconfig cv1800b_milkv_duo_sd
 Run  function
_call_kconfig_script:11: no such file or directory: /home/freire/duo-buildroot-sdk/build/scripts/.py
```
Any suggestions on how to solve it?

micropythondevel, 2024-03-01:
br-rootfs-pack sigue fallando

xpww, 2024-03-01:
In the end, I had no choice, so I used the docker image to compile

Bender1983, 2024-03-28:
there is also another post to this problem, see https://github.com/milkv-duo/duo-buildroot-sdk/issues/74

MarkMLl, 2024-04-08:
I've been chewing at this for a few days, and there appear to be two distinct problems.

Context: host is Debian 12 "Bookworm", build environment ditto running under UML.

The first is that python-lxml will not build, so as an interim measure remove it from package/Config.in (and possibly from package/Config.in.host, in which case the directory may also be removed "just in case".

The second is that the final xz fails. Trimming the path from the command line and running manually it boils down to

<pre>
$ xz -9 -C crc32 -c -T 2 ~/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/images/rootfs.tar > ~/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/images/rootfs.tar.xz
Killed
</pre>

Removing the -9 option fixes it, replacing that with -z 9 or --compress 9 results in an error which suggests that the compression option is problematic in at least some variants of xz.

I've not yet found what's generating this, but there's suspiciously-similar code in buildroot-2021.05/Makefile which something might be clandestinely converting to an invocation of xz.

<pre>
        tar rf $(OUT).tar $(OUT)
        gzip -9 -c < $(OUT).tar > $(OUT).tar.gz
        bzip2 -9 -c < $(OUT).tar > $(OUT).tar.bz2
        rm -rf $(OUT) $(OUT).tar
</pre>

Considering the extent to which xz has been in the news recently I find a hidden operation of this nature troubling.

MarkMLl, 2024-04-09:
So assuming v1.0.9 of duo-buildroot-sdk-develop:

In buildroot-2021.05/package/Config.in remove the line

`"package/python-lxml/Config.in"`

In buildroot-2021.05/fs/common.mk remove the -9 option from

`ROOTFS_$(2)_COMPRESS_CMD = xz -9 -C crc32 -c`

In device/gen_burn_image_sd.sh add host paths:

<pre>
genimage --mkdosfs /usr/sbin/mkdosfs --tune2fs /usr/sbin/tune2fs --e2fsck /usr/sbin/e2fsck \
    --config ${TOP_DIR}/device/${MV_BOARD}/genimage.cfg --rootpath fs/ --inputpath ${PWD} --outputpath ${PWD}
</pre>

Then build using ./build.sh milkv-duo as documented, which eventually produces a rewarding

`Create SD image successful: out/milkv-duo-20240409-1235.img`

...which I can confirm boots and runs.

While the unimplemented option to xz appears fairly uncontroversial, a proper fix for the python-lxml package and incorporating the correct host paths into device/gen_burn_image_sd.sh are "left as an exercise".