milkv-duo/duo-buildroot-sdk #64: Mounting DATA partition Error

thiencuong, 2024-01-27:
Hello.
I meet issue when Mounting DATA partition. Please help

Log: [kernel boot error.log](https://github.com/milkv-duo/duo-buildroot-sdk/files/14071169/kernel.boot.error.log)
Mounting DATA partition
[    3.117185] ubi3: attaching mtd7
[    3.124463] ubi3 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 6:0, read only 0 bytes, retry
[    3.136230] ubi3 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 6:0, read only 0 bytes, retry
[    3.148351] ubi3 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 6:0, read only 0 bytes, retry
[    3.160337] ubi3 error: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 6:0, read 0 bytes
[    3.170804] CPU: 0 PID: 153 Comm: ubiattach Not tainted 5.10.4-tag- #1
[    3.177831] Call Trace:
[    3.180662] [<ffffffe00022c078>] walk_stackframe+0x0/0x6c
[    3.186546] [<ffffffe0004a3e5c>] ubi_io_read+0x1ea/0x2c6
[    3.192344] [<ffffffe0004a4140>] ubi_io_read_ec_hdr+0x8a/0x24e
[    3.198679] [<ffffffe0004a91c4>] ubi_attach+0x2bc/0xefa
[    3.204393] [<ffffffe000295078>] __vmalloc_node_range+0xf8/0x17a
[    3.210905] [<ffffffe00049ef46>] ubi_attach_mtd_dev+0x78c/0xb18
[    3.217327] [<ffffffe00049f8fc>] ctrl_cdev_ioctl+0xbc/0x19c
[    3.223397] [<ffffffe0002ac5dc>] ioctl_file_clone+0x7c/0x80
[    3.229464] [<ffffffe0002ac1f8>] vfs_ioctl+0xc/0x22
[    3.234808] [<ffffffe0002ac5dc>] ioctl_file_clone+0x7c/0x80
[    3.240869] [<ffffffe0002ac68e>] sys_ioctl+0xae/0x538
[    3.246396] [<ffffffe0002ac5dc>] ioctl_file_clone+0x7c/0x80
[    3.252470] [<ffffffe00022b05a>] check_syscall_nr+0x1e/0x22
[    3.258529] UBI assert failed in ubi_io_read at 190 (pid 153)
[    3.264757] CPU: 0 PID: 153 Comm: ubiattach Not tainted 5.10.4-tag- #1
[    3.271784] Call Trace:
[    3.274340] [<ffffffe00022c078>] walk_stackframe+0x0/0x6c
[    3.280328] [<ffffffe0004a3e92>] ubi_io_read+0x220/0x2c6
[    3.286127] [<ffffffe0004a4140>] ubi_io_read_ec_hdr+0x8a/0x24e
[    3.292464] [<ffffffe0004a91c4>] ubi_attach+0x2bc/0xefa
[    3.298172] [<ffffffe000295078>] __vmalloc_node_range+0xf8/0x17a
[    3.304687] [<ffffffe00049ef46>] ubi_attach_mtd_dev+0x78c/0xb18
[    3.311148] [<ffffffe00049f8fc>] ctrl_cdev_ioctl+0xbc/0x19c
[    3.317246] [<ffffffe0002ac5dc>] ioctl_file_clone+0x7c/0x80
[    3.323348] [<ffffffe0002ac1f8>] vfs_ioctl+0xc/0x22
[    3.328703] [<ffffffe0002ac5dc>] ioctl_file_clone+0x7c/0x80
[    3.334769] [<ffffffe0002ac68e>] sys_ioctl+0xae/0x538
[    3.340300] [<ffffffe0002ac5dc>] ioctl_file_clone+0x7c/0x80
[    3.346366] [<ffffffe00022b05a>] check_syscall_nr+0x1e/0x22
[    3.352450] ubi3 error: ubi_attach_mtd_dev: failed to attach mtd7, error -5
ubiattach: error!: cannot attach mtd7
           error 5 (I/O error)
ubiformat: mtd7 (nand), size 5505024 bytes (5.2 MiB), 42 eraseblocks of 131072 bytes (128.0 KiB), min. I/O size 2048 bytes
libscan: scanning eraseblock 6 -- 16 % complete

yue-xiaomin, 2024-01-27:
Mounting data requires a blank directory at the mounting point, and if there is content in that directory, the mounted content is not visible. This section of the mounting script is automatically generated based on the XML layout file. It feels like there are remnants of history. It is recommended to pull and compile the SDK again.

thiencuong, 2024-01-28:
> Mounting data requires a blank directory at the mounting point, and if there is content in that directory, the mounted content is not visible. This section of the mounting script is automatically [...]

I already pull the latest but it's still happen.

thiencuong, 2024-01-29:
it's worked.
Thank you

closed 2024-01-29 by thiencuong