milkv-duo/duo-buildroot-sdk #68: Lack of an arm build option

AmirDahan, 2024-02-04:
Milk V Duo (256mb) has an arm core on it, However there is no info about it in the documentation,
Nor any build options (as far as i'm aware) in this repo.

Is there any way to compile things for arm instead of riscv?

orangecms, 2024-05-11:
There is. I have written code for it and run it, will push that at some point.
Just use an aarch64 toolchain. 
The initial base address in SRAM is different. You can find it buried in the FSBL code.

Here are my notes on running initial code over USB:
https://github.com/orangecms/oreboot/tree/vf2-xtask/src/mainboard/milk-v/duo-s

And some logs, including from running on the Arm core:
https://gist.github.com/orangecms/6df06421cace3d8d1644de583983b826

orangecms, 2024-05-12:
Alright, code is here now:
https://github.com/orangecms/oreboot/tree/vf2-xtask/src/mainboard/milk-v/duo-s/bt0a

Just enter `make run` in there. With `rustup` set up, the `rust-toolchain.toml` in the oreboot repo will automatically get the aarch64 toolchain downloaded. I'm currently using a quick-and-dirty solution via `aarch64-linux-gnu-objcopy` to create the raw binary. 

You will need `sg_boot` installed as per the [main README](https://github.com/orangecms/oreboot/blob/vf2-xtask/src/mainboard/milk-v/duo-s/README.md).

And to switch to the Arm core and enter the USB loader mode, connect pin 35 (boot switch) to GND and pin 32 (GP27) to 3.3V. I'm using a little resistor for each. See also:
https://milkv.io/docs/duo/getting-started/duo256m#risc-v-and-arm-switching

System64fumo, 2024-05-12:
Not exactly what i meant, I meant booting linux/android using the arm core.
I'm sure it's possible to boot linux on the duo by manually building U-Boot, Kernel, Buildroot, ect
But as far as i can see there's no easy way to pick ARM instead of risc-v.

orangecms, 2024-05-12:
That is true, the vendor hasn't prepared that.

OTOH, you could give it a shot. I'd recommend dropping Buildroot and U-Boot entirely to make it simpler.
Just do the DRAM init, load Arm TF-A, and jump to Linux. And use a small rootfs, e.g  a u-root initramfs. We have prebuilt ones in a few flavors even.