milkv-duo/duo-buildroot-sdk #69: How to change the root partition size of Duo 256M?

shiptux, 2024-02-07:


thiagohdaqw, 2024-02-09:
The easiest way I found was to put it on the PC and resize it with gparted

shiptux, 2024-02-20:
> The easiest way I found was to put it on the PC and resize it with gparted

Yeah. I use fdisk to resize the rootfs. But I can also use "milkv/genimage-milkv-duo.cfg" file to change the default size of the root partition in the last version. This file seems be deleted.

carbonfix, 2024-02-20:
Duo:            device/milkv-duo/genimage.cfg
Duo256M:   device/milkv-duo256m/genimage.cfg
DuoS:          device/milkv-duos-sd/genimage.cfg

shiptux, 2024-02-20:
> The easiest way I found was to put it on the PC and resize it with gparted

> Duo: device/milkv-duo/genimage.cfg Duo256M: device/milkv-duo256m/genimage.cfg DuoS: device/milkv-duos-sd/genimage.cfg

Thank you very much.

closed 2024-02-20 by shiptux