milkv-duo/duo-buildroot-sdk #70: camera_test.sh crashes immediately on milkv-dup 256m

cmirceac, 2024-02-18:
Hello, both on individual board and using the carrier board the camera_test.sh starts and then crashes immediately with similar messages, both using the compiled image and the binary release milkv-duo256m-v1.0.8-2024-0130.img, please help:

[root@milkv-duo]~# camera-test.sh 
[SAMPLE_COMM_SNS_ParseIni]-1950: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1605: devNum =  1
[parse_sensor_name]-1686: sensor =  GCORE_GC2083_MIPI_2M_30FPS_10BIT
[parse_sensor_busid]-1714: bus_id =  2
[parse_sensor_i2caddr]-1725: sns_i2c_addr =  37
[parse_sensor_mipidev]-1736: mipi_dev =  0
[parse_sensor_laneid]-1747: Lane_id =  1, 0, 2, -1, -1
[parse_sensor_pnswap]-1758: pn_swap =  0, 0, 0, 0, 0
MMF Version:7e0cc6a08-musl_riscv64
Create VBPool[0], size: (3110400 * 3) = 9331200 bytes
Create VBPool[1], size: (3110400 * 3) = 9331200 bytes
Create VBPool[2], size: (2764800 * 1) = 2764800 bytes
Total memory of VB pool: 21427200 bytes
Initialize SYS and VB
Initialize VI
ISP Vipipe(0) Allocate pa(0x8c771000) va(0x0x3fe736d000) size(291120)
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 25.000000 wdrMode 0 pstSnsObj 0x3fe7ea2400
[SAMPLE_COMM_VI_StartMIPI]-483: sensor 0 stDevAttr.devno 0
awbInit ver 6.8@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
ViPipe:0,===GC2083 1080P 30fps 10bit LINE Init OK!===
********************************************************************************
cvi_bin_isp message
gerritId:      36403          commitId:      c69c5863e      
md5:           cab880835a2ad5184de5ed7762404b84
sensorNum      1              
sensorName0    2083           

PQBIN message
gerritId:      80171          commitId:      5c9d8fc5d      
md5:           ba5a510e093ad42db6788e6c2d13169e
sensorNum      3              
sensorName0    2053           

author:        wanqiang.he    desc:          思博慧CV1812H_GC2083_RGB_mode_V1.0.0
createTime:    2023-08-04 16:48:08version:       V1.1           
tool Version:       v3.0.5.24           mode:      
********************************************************************************
sensorName(0) mismatch, mwSns:2083 != pqBinSns:2053
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
Initialize VPSS
---------VPSS[0]---------
Input size: (1920x1080)
Input format: (19)
VPSS physical device number: 1
Src Frame Rate: -1
Dst Frame Rate: -1
    --------CHN[0]-------
    Output size: (1920x1080)
    Depth: 1
    Do normalization: 0
        Src Frame Rate: -1
        Dst Frame Rate: -1
    ----------------------
    --------CHN[1]-------
    Output size: (1920x1080)
    Depth: 1
    Do normalization: 0
        Src Frame Rate: -1
        Dst Frame Rate: -1
    ----------------------
------------------------
Bind VI with VPSS Grp(0), Chn(0)
Attach VBPool(0) to VPSS Grp(0) Chn(0)
Attach VBPool(1) to VPSS Grp(0) Chn(1)
Initialize VENC
venc codec: h264
venc frame size: 1920x1080
Initialize RTSP
rtsp://10.42.0.216/h264
prio:0
version: 1.4.0
scrfd768432 Build at 2023-12-25 01:21:44 For platform cv181x
Max SharedMem size:1658880
anchor:-8,-8,8,8
anchor:-16,-16,16,16
bbox:bbox_8_Conv_dequant
landmark:kps_8_Conv_dequant
score:score_8_Sigmoid_dequant
anchor:-32,-32,32,32
anchor:-64,-64,64,64
bbox:bbox_16_Conv_dequant
landmark:kps_16_Conv_dequant
score:score_16_Sigmoid_dequant
anchor:-128,-128,128,128
anchor:-256,-256,256,256
bbox:bbox_32_Conv_dequant
landmark:kps_32_Conv_dequant
score:score_32_Sigmoid_dequant
Enter TDL thread
Enter encoder thread
0 R:1165 B:3087 CT:2688
1 R:1464 B:2327 CT:3937
2 R:1974 B:1613 CT:7225
Golden 1464 1024 2327
wdrLEOnly:1
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
face count: 0
[  451.777711] Unable to handle kernel NULL pointer dereference at virtual address 00000000000002a2
[  451.786796] Oops [#1]
[  451.789142] Modules linked in: cv181x_pwm(FO) cv181x_ive(FO) cvi_vc_driver(FO) cv181x_jpeg(FO) cv181x_vcodec(FO) cv181x_tpu(FO) cv181x_clock_cooling(FO) cv181x_rgn(FO) cv181x_mipi_tx(FO) cv181x_vo(FO) cv181x_dwa(FO) cv181x_vpss(FO) cv181x_vi(FO) snsr_i2c(FO) cvi_mipi_rx(FO) cv1)
[  451.817022] CPU: 0 PID: 0 Comm: swapper Tainted: GF          O      5.10.4-tag- #1
[  451.824839] epc: 00000000000002a2 ra : ffffffe00061fb8c sp : ffffffe0007eff10
[  451.832207]  gp : ffffffe0008632c0 tp : ffffffe0007f4440 t0 : 0000000000000000
[  451.839665]  t1 : ffffffe000620798 t2 : 0000000000000000 s0 : ffffffe0007eff70
[  451.847122]  s1 : ffffffe0007f4440 a0 : 0000000000000000 a1 : 0000000000000000
[  451.854580]  a2 : 0000000000000000 a3 : 0000000000000001 a4 : ffffffffffffffff
[  451.862037]  a5 : 0000000000000000 a6 : 0000000000000000 a7 : 0000000054494d45
[  451.869494]  s2 : 0000000000000000 s3 : ffffffe0007fbdc0 s4 : ffffffe0008640a8
[  451.876952]  s5 : 0000000000000001 s6 : 0000000000000000 s7 : 0000000000000000
[  451.884409]  s8 : 000000008b0c90b2 s9 : 0000000081ae02f8 s10: 0000000000000000
[  451.891866]  s11: 0000000000000003 t3 : 0000000000000000 t4 : 0000000000000150
[  451.899322]  t5 : 0000003fe6c89d80 t6 : 0000000000040000
[  451.904808] status: 0000000200000100 badaddr: 00000000000002a2 cause: 000000000000000c
[  451.912980] Call Trace:
[  451.915530] [<ffffffe000620092>] schedule_idle+0xe/0x22
[  451.920935] [<ffffffe0002448ac>] cpu_startup_entry+0x8/0xe
[  451.926607] [<ffffffe000202938>] start_kernel+0x3f2/0x41c
[  451.932211] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=49
[  451.938846] cvi_vip_job_finish:844(): ****img(1) grp(0) finish****
[  451.945852] ---[ end trace 8c9141ee810fbd9b ]---
[  451.950657] Kernel panic - not syncing: Attempted to kill the idle task!
[  451.957585] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

piotr-defendeye, 2024-03-05:
Can confirm, same thing happens to me on the new 1.0.9 release, no matter if downloaded from the official releases or compiled myself.

cmirceac, 2024-03-05:
the question is: is there any image that actually works ?

On 05.03.24 11:01, piotr-defendeye wrote:
>
> Can confirm, same thing happens to me on the new 1.0.9 release, no 
> matter if downloaded from the official releases or compiled myself.
>
> Message ID: ***@***.***>
>

carbonfix, 2024-03-07:
We are working on fixing this issue and will reply here if there is an update

piotr-defendeye, 2024-03-19:
@carbonfix Any updates?

carbonfix, 2024-04-15:
Please use the latest [tdl-sdk](https://github.com/milkv-duo/cvitek-tdl-sdk-sg200x) to compile `sample_vi_fd`, replace `/mnt/system/usr/bin/sample_vi_fd` in Duo256M, and then run `camera-test.sh` to try.

XeL64, 2024-04-18:
> Please use the latest [tdl-sdk](https://github.com/milkv-duo/cvitek-tdl-sdk-sg200x) to compile `sample_vi_fd`, replace `/mnt/system/usr/bin/sample_vi_fd` in Duo256M, and then run `camera-test.sh` to try.

Hello, just try to compile new sample_vi_fd and nothing change, cameta-test.sh start and few second after kernel panic

new sample_vi_fd : 
```
[24-04-18 17:07:31] % file cvi_tdl/sample_vi_fd 
cvi_tdl/sample_vi_fd: ELF 64-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-riscv64v0p7_xthead.so.1, stripped
```
The kernel panic :
```
Starting app...

[root@milkv-duo]~# [   39.790633] Unable to handle kernel paging request at virtual address ffffffdfffe425b0
[   39.798818] Oops [#1]
[   39.801165] Modules linked in: cv181x_pwm(FO) cv181x_ive(FO) cvi_vc_driver(FO) cv181x_jpeg(FO) cv181x_vcodec(FO) cv181)
[   39.827344] CPU: 0 PID: 0 Comm: swapper Tainted: GF          O      5.10.4-tag- #1
[   39.835161] epc: ffffffdfffe425b0 ra : ffffffe000626b98 sp : ffffffe0007fbf10
[   39.842529]  gp : ffffffe00086f920 tp : ffffffe000800440 t0 : 0000000000000000
[   39.849986]  t1 : 0000000001806000 t2 : 0000000000000000 s0 : ffffffe0007fbf70
[   39.857444]  s1 : ffffffe000800440 a0 : ffffffe002200000 a1 : ffffffe000800440
[   39.864901]  a2 : 00000000000007f0 a3 : 0000000000000000 a4 : 0000000000000000
[   39.872359]  a5 : 000000008b0c90b2 a6 : 0000000000000002 a7 : ffffffe000807e88
[   39.879817]  s2 : ffffffe002200000 s3 : ffffffe002228000 s4 : ffffffe0008710a8
[   39.887274]  s5 : 0000000000004000 s6 : ffffffe000800838 s7 : ffffffe0007a0420
[   39.894731]  s8 : ffffffe000807e00 s9 : 0000000081ae5a1c s10: 0000000000000000
[   39.902189]  s11: 0000000000000003 t3 : 0000000000000000 t4 : 7fffffffffffffff
[   39.909645]  t5 : 0000000000000009 t6 : 0000000000040000
[   39.915131] status: 0000000200000100 badaddr: ffffffdfffe425b0 cause: 000000000000000c
[   39.923303] Call Trace:
[   39.925850] [<ffffffe000626eb2>] schedule_idle+0xe/0x22
[   39.931254] [<ffffffe0002448ac>] cpu_startup_entry+0x8/0xe
[   39.936926] [<ffffffe000202938>] start_kernel+0x3f2/0x41c
[   39.942529] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=270
[   39.949241] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=271
[   39.955931] cvi_vip_job_finish:844(): ****img(1) grp(0) finish****
[   39.963017] ---[ end trace a4f34d4289e2d4c7 ]---
[   39.967826] Kernel panic - not syncing: Attempted to kill the idle task!
[   39.974754] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---
```

XeL64, 2024-04-18:
> For information I use v1.1.0 : https://github.com/milkv-duo/duo-buildroot-sdk/releases/download/Duo-V1.1.0/milkv-duo256m-v1.1.0-2024-0410.img.zip
> 
> And I use IO board and just set fixed mac address nothing more.
> 
> ```
> [root@milkv-duo]~# uname -a
> Linux milkv-duo 5.10.4-tag- #1 PREEMPT Wed Apr 10 21:37:02 CST 2024 riscv64 GNU/Linux
> ```
> 
> ```
> [root@milkv-duo]~# file sample_vi_fd 
> sample_vi_fd: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-riscv64v0p7_xthead.so.1, stripped
> 
> [root@milkv-duo]~# ls -halFG /lib/ld-musl-riscv64v0p7_xthead.so.1
> lrwxrwxrwx 1 root 37 Feb 26 11:07 /lib/ld-musl-riscv64v0p7_xthead.so.1 -> ../usr/lib64v0p7_xthead/lp64d/libc.so*
> 
> [root@milkv-duo]~# file /usr/lib64v0p7_xthead/lp64d/libc.so 
> /usr/lib64v0p7_xthead/lp64d/libc.so: ELF 64-bit LSB shared object, UCB RISC-V, version 1 (SYSV), statically linked, stripped
> ```

sample_vi_fd was copied in the right place :
```
[root@milkv-duo]~# file /mnt/system/usr/bin/sample_vi_fd
/mnt/system/usr/bin/sample_vi_fd: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-riscv64v0p7_xthead.so.1, stripped
[root@milkv-duo]~# ls -halFG /mnt/system/usr/bin/sample_vi_fd
-rwxr-xr-x 1 root 35K Apr 18 15:10 /mnt/system/usr/bin/sample_vi_fd*
```

piotr-defendeye, 2024-04-23:
Can confirm, the example still doesn't work. Both camera-test.sh and sample_vi_fd compiled from the newest TDL sdk crash. @carbonfix can we get a fix for this? it's been two months already

teknoman117, 2024-05-14:
@carbonfix looks like this is still an issue on 1.1.0 and the Duo 256M

`[root@milkv-duo]~/cvi_tdl# ./sample_vi_fd /mnt/cvimodel/scrfd_768_432_int8_1x.cv
imodel
[SAMPLE_COMM_SNS_ParseIni]-1950: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1605: devNum =  1
[parse_sensor_name]-1686: sensor =  GCORE_GC2083_MIPI_2M_30FPS_10BIT
[parse_sensor_busid]-1714: bus_id =  2
[parse_sensor_i2caddr]-1725: sns_i2c_addr =  37
[parse_sensor_mipidev]-1736: mipi_dev =  0
[parse_sensor_laneid]-1747: Lane_id =  1, 0, 2, -1, -1
[parse_sensor_pnswap]-1758: pn_swap =  0, 0, 0, 0, 0
MMF Version:7e0cc6a08-musl_riscv64
Create VBPool[0], size: (3110400 * 5) = 15552000 bytes
Create VBPool[1], size: (1382400 * 5) = 6912000 bytes
Create VBPool[2], size: (2764800 * 3) = 8294400 bytes
Total memory of VB pool: 30758400 bytes
Initialize SYS and VB
Initialize VI
ISP Vipipe(0) Allocate pa(0x8d056000) va(0x0x3fc51d9000) size(291120)
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 25.000000 wdrMode 0 pstSnsObj 0x3fc602b878
[SAMPLE_COMM_VI_StartMIPI]-483: sensor 0 stDevAttr.devno 0
awbInit ver 6.8@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
ViPipe:0,===GC2083 1080P 30fps 10bit LINE Init OK!===
********************************************************************************
cvi_bin_isp message
gerritId:      36403          commitId:      c69c5863e
md5:           cab880835a2ad5184de5ed7762404b84
sensorNum      1
sensorName0    2083

PQBIN message
gerritId:      80171          commitId:      5c9d8fc5d
md5:           ba5a510e093ad42db6788e6c2d13169e
sensorNum      3
sensorName0    2053

author:        wanqiang.he    desc:          思博慧CV1812H_GC2083_RGB_mode_V1.0.0
createTime:    2023-08-04 16:48:08version:       V1.1
tool Version:       v3.0.5.24           mode:
********************************************************************************
sensorName(0) mismatch, mwSns:2083 != pqBinSns:2053
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
Initialize VPSS
---------VPSS[0]---------
Input size: (1920x1080)
Input format: (19)
VPSS physical device number: 1
Src Frame Rate: -1
Dst Frame Rate: -1
    --------CHN[0]-------
    Output size: (1280x720)
    Depth: 1
    Do normalization: 0
        Src Frame Rate: -1
        Dst Frame Rate: -1
    ----------------------
    --------CHN[1]-------
    Output size: (1280x720)
    Depth: 1
    Do normalization: 0
        Src Frame Rate: -1
        Dst Frame Rate: -1
    ----------------------
[... 491 lines cut ...]
[  113.981905] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  113.988183] vpss_handle_online_frame_done:2439(): grp(0) chn(0) end
[  113.994681] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=137
[  113.994698] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  114.008579] vpss_handle_online_frame_done:2439(): grp(0) chn(1) end
[  114.022653] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  114.029070] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=138
[  114.029122] cvi_vip_job_finish:844(): ****img(1) grp(0) finish****
[  114.042255] cvi_img_device_run:437(): ****img(1) grp(0) start****
[  114.048558] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  114.054836] vpss_handle_online_frame_done:2439(): grp(0) chn(0) end
[  114.061332] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=139
[  114.061349] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  114.075220] vpss_handle_online_frame_done:2439(): grp(0) chn(1) end
[  114.089301] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  114.095717] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=140
[  114.095770] cvi_vip_job_finish:844(): ****img(1) grp(0) finish****
[  114.108929] cvi_img_device_run:437(): ****img(1) grp(0) start****
[  114.115250] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  114.121521] vpss_handle_online_frame_done:2439(): grp(0) chn(0) end
[  114.128016] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=141
[  114.128033] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  114.141917] vpss_handle_online_frame_done:2439(): grp(0) chn(1) end
[  114.154823] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: unmap_page_range+0x1bc/0x30c
[  114.165616] ---[ end Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: unmap_page_range+0x1bc/0x30c ]---`

carbonfix, 2024-05-15:
We have reproduced this problem and have not yet found the root cause. We have asked the CPU manufacturer to assist in analysis.

carbonfix, 2024-05-29:
Please try using firmware V1.1.1 or later.
https://github.com/milkv-duo/duo-buildroot-sdk/releases

teknoman117, 2024-05-29:
@carbonfix Seems to be working for me now. What did disabling "TPU_PERF_MODE" do? Does the NPU still run faster than the one in the regular Duo (the 64M one) and the Duo S?

teknoman117, 2024-05-29:
Carbonfix answered the question on the forums - this halves the performance of the TPU. But no more crashes. I'm guessing it's the same hardware issue the LicheeRV is dealing with (they noted that their "beta boards", which is all they're selling, only offer 0.5 TOP/s versus 1.0 TOPs with the SG2002).

cmirceac, 2024-06-01:
As I have originally opened this issue I can now confirm that the script doesn't crash the kernel anymore and it even recognize my face in **text mode**, the messages that a face has been recognized appear in the scrolling log, but the RTSP "server" doesn't seem to work at all, either with VLC or Celluloid player on the latest Linux Mint, I will close this issue and open another one related to the video streaming.

closed 2024-06-01 by cmirceac