milkv-duo/duo-buildroot-sdk #74: make: *** [Makefile:605: br-rootfs-pack] Error 2

micropythondevel, 2024-03-09:
2024-03-09T20:23:57 >>> linux-firmware 20201022 Installing to images directory
2024-03-09T20:23:58 >>> subversion 1.14.1 Installing to target
Done in 14min 01s  (error code: 2)
make: *** [Makefile:605: br-rootfs-pack] Error 2
Error: Build board milkv-duo failed!
root@74ef18797463:/home/work#

SpiroCx, 2024-03-17:
I had the same error compiling on Ubuntu 22.04.  All of these failed:
build.sh milkv-duo, build.sh milkv-duo-light, step-by-step duo/duo-light, docker duo/duo-light.

I tried them all again after checking out tag Duo-V1.0.9 (commit c737560cc).  Same errors, so I checked out develop again (commit 1d878b3d9).

I decided to focus on the docker build.  I don't know what the proper solution is but I got it going.  
The main way to find the workaround was
- replace that line 605 of build/Makefile (as mentioned in the error message) with "make -j1 -C $(BR_DIR)"
- install strace into the container and use it to take a closer look at "make br-rootfs-pack"

Removing the brmake wrapper (brmake -> make) and running make with `make SHELL='sh -x'` give more info.  This shows there a problem with a fakeroot call.

You can verify this path to fakeroot.so problem by running the fakeroot line separately and adding strace:
```
export PATH="/home/work/buildroot-2021.05/output/milkv-duo-lite_musl_riscv64/host/bin:/home/work/buildroot-2021.05/output/milkv-duo-lite_musl_riscv64/host/sbin:/home/work/host-tools/gcc/riscv64-linux-musl-x86_64/bin:/home/work/host-tools/gcc/riscv64-linux-x86_64/bin:/home/work/host-tools/gcc/riscv64-elf-x86_64/bin:/home/work/host-tools/gcc/gcc-linaro-6.3.1-2017.05-x86_64_aarch64-elf/bin:/home/work/host-tools/gcc/gcc-linaro-6.3.1-2017.05-x86_64_arm-linux-gnueabihf/bin:/home/work/host-tools/gcc/gcc-linaro-6.3.1-2017.05-x86_64_aarch64-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/work/host-tools/gcc/arm-cvitek-linux-uclibcgnueabihf/bin"
export FAKEROOTDONTTRYCHOWN=1
strace /home/work/buildroot-2021.05/output/milkv-duo-lite_musl_riscv64/host/bin/fakeroot -- /home/work/buildroot-2021.05/output/milkv-duo-lite_musl_riscv64/build/buildroot-fs/ext2/fakeroot 2>&1 | grep libfakeroot.so 
```

In the docker container, at least, the problem is that the external paths are being used inside the container.  I needed to do this to get the docker build working:

##### replicate base dir inside docker container (`docker exec -it duodocker /bin/bash`):
```
mkdir -p /home/<host path>/duo-buildroot-sdk/
cd /home/<host path>/duo-buildroot-sdk/
ln -s /home/work/buildroot-2021.05 .
ln -s /home/work/host-tools .
```

NOTE: I also did these 2 things in the container which may be useless, or may be significant:
    `apt install kmod`
    `apt remove fakeroot`
- the first was because I noticed an error message regarding missing util depmod
- the second was because there is a fakemod in the sdk and one of my guesses was that it may be using the wrong one.

I can confirm now that both of these commands work outside the docker container (running on the host machine):
```
docker exec -it duodocker /bin/bash -c "cd /home/work && cat /etc/issue && ./build.sh milkv-duo-lite"
docker exec -it duodocker /bin/bash -c "cd /home/work && cat /etc/issue && ./build.sh milkv-duo"
``` 

and you get your images in: /duo-buildroot-sdk/out/
- milkv-duo-lite.img
- milkv-duo.img

Bender1983, 2024-03-20:
Had the same problem...
After entering  buildroot-2021.05 and executing make it was easy to fix the error (NEWLINE)
Fixed in  buildroot-2021.05/support/dependencies/dependencies.mk

regards

MarkMLl, 2024-03-28:
Could we have a bit more detail on your dependencies.mk fix please? I'm new to buildroot so am struggling slightly with the format.

informatik-bks, 2024-03-28:
after executing make in the directory it was easy to fix the error in the dependencies.mk =>NEWLINE

Bender1983, 2024-03-28:
there is also another post to this problem, see https://github.com/milkv-duo/duo-buildroot-sdk/issues/62

MarkMLl, 2024-03-28:
> ps: sorry for the strange format

So do you mean like this, with the newline replaced by a single space?

`dependencies:`
`        @MAKE="$(MAKE)" DL_TOOLS="$(sort $(DL_TOOLS_DEPENDENCIES))" $(TOPDIR)/support/dependencies/dependencies.sh`

I used backtick quoting on each line to get that and have checked the preview... a browser might display that as three lines, but it should really be two.

I have to go out, so am casually leaving that to run.

MarkMLl, 2024-03-30:
In my case that didn't fix the problem, and looking more closely it's a "catch-all" error that can be caused by any failure in the brmake script that iteratively does all the work: output is redirected to build/br.log where I find

`Could not find function xmlCheckVersion in library libxml2. Is libxml2 installed?`

It looks as though this is being caused by the fact that libxml2 as brought in as part of buildroot is at 2.9.12, while the Debian-based host system already has 2.9.14 which for some reason doesn't include that function. I've checked but x86_64 and Arm-based Debian systems and they're the same, looking around I can see xmlCheckVersion() on an older Debian x86_64 system where the library is at 2.9.4.

I'll continue tinkering with this, but it could explain why some people can only build using the Docker image.

MarkMLl, 2024-04-04:
The error is the same irrespective of whether the host is Debian "Bookworm" (which omits xmlCheckVersion() from libxml2) or Bullseye (which has it): it's not so much that libxml2 is missing but that while

<pre>
...
2024-04-04T14:20:34 >>> python-lxml 4.6.3 Extracting
2024-04-04T14:20:35 >>> python-lxml 4.6.3 Patching
2024-04-04T14:20:36 >>> python-lxml 4.6.3 Configuring
2024-04-04T14:20:42 >>> python-lxml 4.6.3 Building
Done in 4h 27min 31s  (error code: 2)
make: *** [Makefile:605: br-rootfs-pack] Error 2
Error: Build board milkv-duo failed!
</pre>

This happens:

<pre>
2024-04-04T14:20:44 creating build/temp.linux-x86_64-3.9/src/lxml
2024-04-04T14:20:44 /home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/bin/riscv64-unknown-linux-musl-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -O3 -Wall -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Os -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Os -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../usr/include/libxml2 -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../../../riscv64-buildroot-linux-musl/sysroot/usr/bin/../../usr/include/libxml2 -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../usr/include -Isrc -Isrc/lxml/includes -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/include/python3.9 -c src/lxml/etree.c -o build/temp.linux-x86_64-3.9/src/lxml/etree.o -w
2024-04-04T14:20:56 ESC[01mESC[Kriscv64-unknown-linux-musl-gcc:ESC[mESC[K ESC[01;31mESC[Kfatal error: ESC[mESC[KKilled signal terminated program cc1
2024-04-04T14:20:56 compilation terminated.
2024-04-04T14:20:56 Compile failed: command '/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/bin/riscv64-unknown-linux-musl-gcc' failed with exit code 1
2024-04-04T14:20:57 creating tmp
2024-04-04T14:20:57 cc -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../usr/include/libxml2 -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../../../riscv64-buildroot-linux-musl/sysroot/usr/bin/../../usr/include/libxml2 -I/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../usr/include -I/usr/include/libxml2 -c /tmp/xmlXPathInitc126fc6l.c -o tmp/xmlXPathInitc126fc6l.o
2024-04-04T14:20:57 cc tmp/xmlXPathInitc126fc6l.o -L/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../../../riscv64-buildroot-linux-musl/sysroot/usr/lib -lxml2 -o a.out
2024-04-04T14:20:57 /usr/bin/ld: skipping incompatible /home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../../../riscv64-buildroot-linux-musl/sysroot/usr/lib/libxml2.so when searching for -lxml2
2024-04-04T14:20:57 /usr/bin/ld: cannot find -lxml2: No such file or directory
2024-04-04T14:20:57 /usr/bin/ld: skipping incompatible /home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../../../riscv64-buildroot-linux-musl/sysroot/usr/lib/libxml2.so when searching for -lxml2
2024-04-04T14:20:57 collect2: error: ld returned 1 exit status
</pre>

despite

<pre>
/home/markMLl/duo-buildroot-sdk-develop/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/python-lxml/host/riscv64-buildroot-linux-musl/sysroot/usr/bin/../../../../riscv64-buildroot-linux-musl/sysroot/usr/lib/libxml2.so -> libxml2.so.2.9.12

ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           RISC-V
  Version:                           0x1
  Entry point address:               0x313c0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1394344 (bytes into file)
  Flags:                             0x5, RVC, double-float ABI
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         5
  Size of section headers:           64 (bytes)
  Number of section headers:         34
  Section header string table index: 33
</pre>

MarkMLl, 2024-04-09:
There are definite problems relating to building the MilkV-Duo, but since most (i.e. two out of three) relate to BuildRoot I will append my fixes here.

Assuming v1.0.9 of duo-buildroot-sdk-develop:

In buildroot-2021.05/package/Config.in remove the line

`"package/python-lxml/Config.in"`

In buildroot-2021.05/fs/common.mk remove the -9 option from

`ROOTFS_$(2)_COMPRESS_CMD = xz -9 -C crc32 -c`

In device/gen_burn_image_sd.sh add host paths:

<pre>
genimage --mkdosfs /usr/sbin/mkdosfs --tune2fs /usr/sbin/tune2fs --e2fsck /usr/sbin/e2fsck \
    --config ${TOP_DIR}/device/${MV_BOARD}/genimage.cfg --rootpath fs/ --inputpath ${PWD} --outputpath ${PWD}
</pre>

Then build using ./build.sh milkv-duo as documented, which eventually produces a rewarding

`Create SD image successful: out/milkv-duo-20240409-1235.img`

...which I can confirm boots and runs.

While the unimplemented option to xz appears fairly uncontroversial, a proper fix for the python-lxml package and incorporating the correct host paths into device/gen_burn_image_sd.sh are "left as an exercise".

PJ5168, 2024-06-01:
On WSL2/Ubuntu 22.04,
Fail the build on currnt git clone.

# create partition mounting points and move sytems to read-write partitions
  [TARGET] br-rootfs-pack
make -C /home/pj/duo-buildroot-sdk/buildroot-2021.05 milkv-duo256m_musl_riscv64_defconfig BR2_TOOLCHAIN_EXTERNAL_PATH=/home/pj/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64
make[1]: Entering directory '/home/pj/duo-buildroot-sdk/buildroot-2021.05'
  GEN     /home/pj/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo256m_musl_riscv64/Makefile
/home/pj/duo-buildroot-sdk/buildroot-2021.05/configs/milkv-duo256m_musl_riscv64_defconfig:368:warning: override: reassigning to symbol BR2_PACKAGE_PROVIDES_HOST_OPENSSL
#
# configuration written to /home/pj/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo256m_musl_riscv64/.config
#
make[1]: Leaving directory '/home/pj/duo-buildroot-sdk/buildroot-2021.05'
/home/pj/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j16 -C /home/pj/duo-buildroot-sdk/buildroot-2021.05
Done in 4s  (error code: 2)
make: *** [Makefile:605: br-rootfs-pack] Error 2
Error: Build board milkv-duo256m failed!

LevitatingBusinessMan, 2024-06-17:
I stumbled upon a similar issue.

After replacing `brmake` with `make` on line 605 in `build/Makefile` and running with `SHELL='sh -x'` I managed to get some more debugging output.

For me the issue appears to be with compiling the `kill` utility

```
make[4]: Entering directory '/home/rein/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo256m_musl_riscv64/build/host-util-linux-2.36.2'
  CC       misc-utils/kill.o
misc-utils/kill.c: In function ‘kill_with_timeout’:
misc-utils/kill.c:394:20: error: implicit declaration of function ‘pidfd_open’; did you mean ‘fdopen’? [-Wimplicit-function-declaration]
  394 |         if ((pfd = pidfd_open(ctl->pid, 0)) < 0)
      |                    ^~~~~~~~~~
      |                    fdopen
misc-utils/kill.c:394:20: warning: nested extern declaration of ‘pidfd_open’ [-Wnested-externs]
misc-utils/kill.c:399:13: error: implicit declaration of function ‘pidfd_send_signal’; did you mean ‘SYS_pidfd_send_signal’? [-Wimplicit-function-declaration]
  399 |         if (pidfd_send_signal(pfd, ctl->numsig, &info, 0) < 0)
      |             ^~~~~~~~~~~~~~~~~
      |             SYS_pidfd_send_signal
misc-utils/kill.c:399:13: warning: nested extern declaration of ‘pidfd_send_signal’ [-Wnested-externs]
make[4]: *** [Makefile:9008: misc-utils/kill.o] Error 1
```

zzhelezc, 2024-07-07:
I am having the same problem building inside Docker on an M1 mac. I tried #110 but I still get the error. I want to build from source because I need #102. Any help would be appreciated.

LevitatingBusinessMan, 2024-07-07:
> I am having the same problem building inside Docker on an M1 mac. I tried #110 but I still get the error. I want to build from source because I need #102. Any help would be appreciated.

Without build logs it is hard to identify the actual issue. 
You can edit the Makefile to replace `brmake` with `make` and you should be able to get better logs.

In any case it's weird that it fails with even the docker container.

zzhelezc, 2024-07-08:
Thanks for the prompt response! I'll build with make and take a look at the logs. 

The official release would work for me but from what I can see there isn't one that has `cdc-ncm`. Maybe I am missing something? Latest official release is v1.1.1 on May 28 and `cdc-ncm` was merged Jun 13.

LevitatingBusinessMan, 2024-07-08:
> The official release would work for me but from what I can see there isn't one that has `cdc-ncm`. Maybe I am missing something? Latest official release is v1.1.1 on May 28 and `cdc-ncm` was merged Jun 13.

Oh correct, the releases are a bit out of date.

zzhelezc, 2024-07-08:
It seems that it starts failing when extracting `alsa-utils`:
```
mkdir -p /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/per-package/alsa-utils/target
bzcat /home/work/buildroot-2021.05/dl/alsa-utils/alsa-utils-1.2.4.tar.bz2 | tar --strip-components=1 -C /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/build/alsa-utils-1.2.4   -xf -
tar: tests/acgo.at: Cannot open: Permission denied
tar: tests/aclang.at: Cannot open: Permission denied
tar: tests/acstatus.at: Cannot open: Permission denied
tar: tests/acc.at: Cannot open: Permission denied
tar: tests/aclibs.at: Cannot open: Permission denied
tar: tests/actypes.at: Cannot open: Permission denied
tar: tests/acfortran.at: Cannot open: Permission denied
tar: tests/acautoupdate.at: Cannot open: Permission denied
tar: tests/acautoheader.at: Cannot open: Permission denied
tar: tests/acheaders.at: Cannot open: Permission denied
tar: tests/acfunctions.at: Cannot open: Permission denied
tar: tests/acprograms.at: Cannot open: Permission denied
tar: tests/acspecific.at: Cannot open: Permission denied
tar: tests/acgeneral.at: Cannot open: Permission denied
tar: t/check2-w.sh: Cannot open: Permission denied
tar: Exiting with failure status due to previous errors
make[1]: *** [package/pkg-generic.mk:175: /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/build/host-autoconf-2.69/.stamp_extracted] Error 2
make[1]: *** Waiting for unfinished jobs....
tar: m4/ltversion.m4: Cannot open: Permission denied
tar: t/compile5-w.sh: Cannot open: Permission denied
tar: t/compile3-w.sh: Cannot open: Permission denied
tar: libltdl/Makefile.am: Cannot open: Permission denied
tar: t/ar-lib-w.sh: Cannot open: Permission denied
tar: t/missing-version-mismatch-w.sh: Cannot open: Permission denied
tar: t/install-sh-option-C-w.sh: Cannot open: Permission denied
tar: t/check6-w.sh: Cannot open: Permission denied
tar: build-aux/ltmain.sh: Cannot open: Permission denied
tar: t/depcomp-lt-msvcmsys.tap: Cannot open: Permission denied
tar: t/mkinst3-w.sh: Cannot open: Permission denied
tar: t/compile4-w.sh: Cannot open: Permission denied
tar: Exiting with failure status due to previous errors
make[1]: *** [package/pkg-generic.mk:175: /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/build/host-libtool-2.4.6/.stamp_extracted] Error 2
tar: t/depcomp-lt-msvisualcpp.tap: Cannot open: Permission denied
tar: t/testsuite-part.am: Cannot open: Permission denied
tar: t/depcomp-lt-makedepend.tap: Cannot open: Permission denied
tar: t/check-tests-in-builddir-w.sh: Cannot open: Permission denied
tar: t/check5-w.sh: Cannot open: Permission denied
tar: t/check-exported-srcdir-w.sh: Cannot open: Permission denied
tar: t/depcomp-auto.tap: Cannot open: Permission denied
tar: t/posixsubst-tests-w.sh: Cannot open: Permission denied
tar: t/compile2-w.sh: Cannot open: Permission denied
tar: t/depcomp-lt-disabled.tap: Cannot open: Permission denied
tar: t/compile7-w.sh: Cannot open: Permission denied
tar: t/maken3-w.sh: Cannot open: Permission denied
tar: t/depcomp-msvisualcpp.tap: Cannot open: Permission denied
tar: t/depcomp-cpp.tap: Cannot open: Permission denied
tar: t/depcomp-disabled.tap: Cannot open: Permission denied
tar: t/mdate6-w.sh: Cannot open: Permission denied
tar: t/exeext4-w.sh: Cannot open: Permission denied
tar: t/depcomp-dashmstdout.tap: Cannot open: Permission denied
tar: t/missing3-w.sh: Cannot open: Permission denied
tar: t/compile-w.sh: Cannot open: Permission denied
tar: t/depcomp-lt-auto.tap: Cannot open: Permission denied
tar: t/check4-w.sh: Cannot open: Permission denied
tar: t/check-fd-redirect-w.sh: Cannot open: Permission denied
tar: t/color-tests-w.sh: Cannot open: Permission denied
tar: t/depcomp-lt-gcc.tap: Cannot open: Permission denied
tar: bootstrap: Cannot open: Permission denied
tar: Exiting with failure status due to previous errors
make[1]: *** [package/pkg-generic.mk:170: /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/build/host-m4-1.4.18/.stamp_extracted] Error 2
tar: t/depcomp-gcc.tap: Cannot open: Permission denied
tar: t/color-tests2-w.sh: Cannot open: Permission denied
tar: t/check-subst-w.sh: Cannot open: Permission denied
tar: t/depcomp-msvcmsys.tap: Cannot open: Permission denied
tar: t/built-sources-check-w.sh: Cannot open: Permission denied
tar: t/check11-w.sh: Cannot open: Permission denied
[... 9 lines cut ...]
tar: t/compile6-w.sh: Cannot open: Permission denied
tar: t/check7-w.sh: Cannot open: Permission denied
tar: Exiting with failure status due to previous errors
make[1]: *** [package/pkg-generic.mk:176: /home/work/buildroot-2021.05/output/milkv-duo_musl_riscv64/build/host-automake-1.15.1/.stamp_extracted] Error 2
make[1]: Leaving directory '/home/work/buildroot-2021.05'
make: *** [Makefile:605: br-rootfs-pack] Error 2
Error: Build board milkv-duo failed!
```

I am a bit confused by all the `permission denied` errors as I am running as root in docker. But maybe I need to pass some other flag to docker? I'll try and debug this further when I have a bit more time.

> You could try using the docker image with [the v2 sdk](https://github.com/milkv-duo/duo-buildroot-sdk-v2)?

It seems that the new sdk only supports the 256m duo and duos.
```
root@bbd7cdf10be0:/home/work# ./build.sh
Usage:
./build.sh              - Show this menu
./build.sh lunch        - Select a board to build
./build.sh [board]      - Build [board] directly, supported boards as follows:
milkv-duo256m-glibc-arm64-sd
milkv-duo256m-musl-riscv64-sd
milkv-duos-glibc-arm64-sd
milkv-duos-musl-riscv64-sd
```

zzhelezc, 2024-07-09:
Digging a bit further, `chmod` was getting all the `Permission denied` errors. A simple (and dirty) `chmod -R 777 duo-buildroot-sdk` outside of docker fixed the problem but the build then failed a bit later. I'll report back when I know what is going on. 

P.S. If anyone can generate a more recent official image (that has `cdc-ncm`) I would be very grateful.

LevitatingBusinessMan, 2024-07-15:
Do you know where it failed? I would generate the image for you but I am on vacation.

zzhelezc, 2024-07-16:
> Do you know where it failed? 

No. I haven't had the time to debug this any further. I'll report back when I have some time to try the build again (the docker build is very slow on an M1 mac)

> I would generate the image for you but I am on vacation.

Thanks for the offer! I am not in a hurry, so please enjoy your vacation.

zzhelezc, 2024-07-28:
I built the image using Docker on a Ubuntu host without any problems! The problems I was encountering on the docker mac build are most likely due to the docker setup.

LevitatingBusinessMan, 2024-08-07:
@zzhelezc interesting, glad you got it to work! 

Though the behaviour of containers should be consistent of their environments, isn't that what containers are for 🤔

zzhelezc, 2024-08-07:
> Though the behaviour of containers should be consistent of their environments, isn't that what containers are for 🤔

I agree that containers should give a reproducible environment independent of the host but that's only true if docker is configured correctly. Given that most of my errors were `Permission denied` I feel like my m1 docker was most likely misconfigured. Do you have to run docker with `sudo`?