milkv-duo/duo-buildroot-sdk #78: setting ION_SIZE = 0 on duo-256m causes build to fail

secretagentmoof, 2024-04-01:
when following the directions in the FAQ, setting ION_SIZE = 0 causes an assertion (ISP_MEM_BASE_ADDR + ISP_MEM_BASE_SIZE <= ION_ADDR + ION_SIZE) to fire and the build to fail
$ git diff
diff --git a/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py b/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
index fcc097d01..ce30e8b90 100755
--- a/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
+++ b/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
@@ -40,7 +40,8 @@ class MemoryMap:
     # =================
     # Multimedia buffer. Used by u-boot/kernel/FreeRTOS
     # =================
-    ION_SIZE = 75 * SIZE_1M
+    #    ION_SIZE = 75 * SIZE_1M
+    ION_SIZE = 0 * SIZE_1M
$ ./build.sh milkv-duo256m
....
  [TARGET] /home/secretagentmoof/duo-buildroot-sdk/build/output/cv1812cp_milkv_duo256m_sd/cvi_board_memmap.h
mkdir -p /home/secretagentmoof/duo-buildroot-sdk/build/output/cv1812cp_milkv_duo256m_sd/
    INFO:mmap_conv: map_file_path is at /home/secretagentmoof/duo-buildroot-sdk/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
Traceback (most recent call last):
  File "/home/secretagentmoof/duo-buildroot-sdk/build/scripts/mmap_conv.py", line 131, in <module>
    main()
  File "/home/secretagentmoof/duo-buildroot-sdk/build/scripts/mmap_conv.py", line 113, in main
    spec.loader.exec_module(mmap_module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/secretagentmoof/duo-buildroot-sdk/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py", line 6, in <module>
    class MemoryMap:
  File "/home/secretagentmoof/duo-buildroot-sdk/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py", line 58, in MemoryMap
    assert ISP_MEM_BASE_ADDR + ISP_MEM_BASE_SIZE <= ION_ADDR + ION_SIZE
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError
make: *** [scripts/mmap.mk:14: /home/secretagentmoof/duo-buildroot-sdk/build/output/cv1812cp_milkv_duo256m_sd/cvi_board_memmap.h] Error 1

carbonfix, 2024-04-01:
Try this:
```diff
diff --git a/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py b/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
index fcc097d01..545af061c 100755
--- a/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
+++ b/build/boards/cv181x/cv1812cp_milkv_duo256m_sd/memmap.py
@@ -40,10 +40,10 @@ class MemoryMap:
     # =================
     # Multimedia buffer. Used by u-boot/kernel/FreeRTOS
     # =================
-    ION_SIZE = 75 * SIZE_1M
-    H26X_BITSTREAM_SIZE = 2 * SIZE_1M
+    ION_SIZE = 0 * SIZE_1M
+    H26X_BITSTREAM_SIZE = 0 * SIZE_1M
     H26X_ENC_BUFF_SIZE = 0
-    ISP_MEM_BASE_SIZE = 20 * SIZE_1M
+    ISP_MEM_BASE_SIZE = 0 * SIZE_1M
     FREERTOS_RESERVED_ION_SIZE = H26X_BITSTREAM_SIZE + H26X_ENC_BUFF_SIZE + ISP_MEM_BASE_SIZE
 
     # ION after FreeRTOS
diff --git a/ramdisk/rootfs/overlay/cv181x_musl_riscv64/system/ko/loadsystemko.sh b/ramdisk/rootfs/overlay/cv181x_musl_riscv64/system/ko/loadsystemko.sh
index b07a47db8..a539510d1 100644
--- a/ramdisk/rootfs/overlay/cv181x_musl_riscv64/system/ko/loadsystemko.sh
+++ b/ramdisk/rootfs/overlay/cv181x_musl_riscv64/system/ko/loadsystemko.sh
@@ -24,7 +24,7 @@ insmod /mnt/system/ko/cv181x_vcodec.ko
 insmod /mnt/system/ko/cv181x_jpeg.ko
 insmod /mnt/system/ko/cvi_vc_driver.ko MaxVencChnNum=9 MaxVdecChnNum=9
 #insmod /mnt/system/ko/cv181x_rtc.ko
-insmod /mnt/system/ko/cv181x_ive.ko
+#insmod /mnt/system/ko/cv181x_ive.ko
 
 echo 3 > /proc/sys/vm/drop_caches
 dmesg -n 4
```

secretagentmoof, 2024-04-03:
I can confirm that with those changes, it successfully builds and `free` reports
```
              total        used        free      shared  buff/cache   available
Mem:         247036       15932      222092          76        9012      224264
```
Could this be added to the FAQ?

carbonfix, 2024-04-03:
Thank you for your report, we will put this part of the description into the documentation, thank you!