milkv-duo/duo-buildroot-sdk #83: build.sh with duo256m is failed

acerols, 2024-04-13:
Hi all,

Now I encounted the below error when I had ran build.sh with milkv-duo256m option.
Does anyone know how to solve this issue?

thank you in advance.

```log
riscv64-unknown-linux-gnu-cpp -DLZ4_USER_MEMORY_FUNCTIONS=1 -DNO_ALLOCS -DARGTYPE=3 -DLTC_NO_FILE -DLTM_NO_FILE -DLTM_DESC -DLTC_SOURCE -DRISCV=/opt/riscv -DTEST_FROM_SPINOR1=0 -DPAGE_SIZE_64KB=1 -DPRINTF_TIMESTAMP=0 -DENABLE_ASSERTIONS=1 -DLOG_LEVEL=2 -D__CVITEK__ -DNANDBOOT_V2=1 -DTOC_HEADER_NAME=0xC906B001 -DRTOS_ENABLE_FREERTOS=y -DRTOS_DUMP_PRINT_SZ_IDX=17 -DRTOS_FAST_IMAGE_TYPE=0 -DRTOS_DUMP_PRINT_ENABLE=1 -DOD_CLK_SEL=y -DFSBL_SECURE_BOOT_SUPPORT=0 -D USB_DL_BY_FSBL -Iinclude -Iinclude/cpu/riscv  -Iplat/ -Iplat/cv181x/include/uart -Iplat/cv181x/include -Iplat/cv181x/include/riscv -Ilib/utils -Ilib/lzma -Ilib/lz4 -Iinclude/stdlib -Iinclude/stdlib/sys -Ilib/libtommath -Ilib/libtomcrypt/src/headers -Ilib/BigDigits -Ibuild -Iinclude/cpu -nostdinc -Wmissing-include-dirs -Werror -P -D__ASSEMBLY__ -D__LINKER__ -Wp,-MD,/home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/blmacros/blmacros.ld.d -MT /home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/blmacros/blmacros.ld -MP -o /home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/blmacros/blmacros.ld make_helpers/get_macros.ld.S
  LD      /home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/blmacros/blmacros.elf
<command-line>: error: "RISCV" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
<command-line>: error: "RISCV" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[1]: *** [Makefile:195: /home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/bl2/bl2_entrypoint.o] Error 1
make[1]: *** Waiting for unfinished jobs....
<command-line>: error: "RISCV" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[1]: *** [Makefile:195: /home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/bl2/bl2_main.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:200: /home/risc-v/duo-buildroot-sdk-develop/fsbl/build/cv1812cp_milkv_duo256m_sd/blmacros/blmacros.elf] Error 1
make[1]: Leaving directory '/home/risc-v/duo-buildroot-sdk-develop/fsbl'
make: *** [scripts/fip_v2.mk:32: fsbl-build] Error 2
Error: Build board milkv-duo256m failed!
```

devemin, 2024-04-18:
I used this and success to build

sudo ./build.sh lunch

acerols, 2024-04-19:
@devemin 
Thanks to your advice, I've complited the build.

closed 2024-04-19 by acerols

zlh20040308, 2024-08-04:
It works! Thank you.

Plaque-fcc, 2025-01-05:
You shouldn't be using `sudo` to *build* this project. Instead, one should locate and fix the problem.

Using `sudo` and especially recommending others to do so left and right without understanding why and what you're achieving with it, it extremely dangerous.