milkv-duo/duo-buildroot-sdk #85: USB Audio Driver Error

devemin, 2024-04-18:
Hello!

Milk-V Duo 256MB
I build kernel, But RNDIS driver is not working. alternatively USB Audio (source & sink )appeared.

Is this commit wrong update ???

https://github.com/milkv-duo/duo-buildroot-sdk/commit/2af594800dfd9628a037351986e969e3b049aa39

Thanks.

[image]

[image]

carbonfix, 2024-04-18:
fixed:
https://github.com/milkv-duo/duo-buildroot-sdk/commit/bee2e35944e38da76326911ed1913823ca8d4164

closed 2024-04-18 by carbonfix

devemin, 2024-04-18:
Thanks,

But the today recent repository can't use RNDIS.

devemin, 2024-04-18:
(pre-build official image can use RNDIS.)

https://milkv.io/ja/docs/duo/getting-started/boot

carbonfix, 2024-04-18:
The RNDIS problem is caused by this USB Audio problem, so use the latest code and it will be normal.