milkv-duo/duo-buildroot-sdk #87: Serial RX (GPIO13) Pulled Down and prevent incoming commands over Serial [Duo256]

hellvesper, 2024-04-23:
I was flashed latest release (and also tried  master build), It boots well, I see console messages and prompt `[root@milkv-duo]~#` but can't type anything. Checked on Logic analyser and realised that UART0_RX / GPIO13 / Pin 17 of Duo 256 board is Pulled Down (should be pulled up) and prevent incoming communication. I think some kind of default config issue. Asking for solution.

Plaque-fcc, 2024-05-31:
Are you sure that GPIO13 is pulled down? I just flashed the v1.1.1 (tag Duo-V1.1.1) image I built recently, and I measure 3V23 on GPIO13 (Pin 17). I didn't reconfigure anything, flashed-booted-measured, that's it.

hellvesper, 2024-05-31:
> Are you sure that GPIO13 is pulled down? I just flashed the v1.1.1 (tag Duo-V1.1.1) image I built recently, and I measure 3V23 on GPIO13 (Pin 17). I didn't reconfigure anything, flashed-booted-measured, that's it.

It could be fixed in this release. But I checked again and by some reason it works, but I'm sure it wasn't, could be floating bug. I'll check with new release