milkv-duo/duo-buildroot-sdk #88: [Duo256] Board by some reason resets to ARM core and hangs up

hellvesper, 2024-04-25:
My setup:
I'm using IO Board. 
Connected Camera and Ethernet cable. 
Boot switch pin not connected. 

I realised that after some time of idle board reset and doesn't boot until reset by button or power. And could boot only after several resets.  Its happens much quickly when in `camera-test` (higher power demand?).  I connect UART and realised that board reset itself and hangs up while trying to boot with ARM core (B.SCS) without reason:

```sh
[    7.652612] cv181x-cooling cv181x_cooling: elems of dev-freqs=6
[    7.658856] cv181x-cooling cv181x_cooling: dev_freqs[0]: 850000000 500000000
[    7.666691] cv181x-cooling cv181x_cooling: dev_freqs[1]: 425000000 375000000
[    7.674279] cv181x-cooling cv181x_cooling: dev_freqs[2]: 425000000 300000000
[    7.681947] cv181x-cooling cv181x_cooling: Cooling device registered: cv181x_cooling
[    7.713723] jpu ctrl reg pa = 0xb030000, va = 00000000bf58e285, size = 256
[    7.721344] end jpu_init result = 0x0
[    7.827102] cvi_vc_drv_init result = 0x0
[    7.892201] sh (184): drop_caches: 3
Starting app...

[root@milkv-duo]~# B.SCS/0/0.WD.URPL.SDI/25000000/6000000.BS/SD.PS.SD/0x0/0x1000/0x1000/0.PE.BS.SD/0x1000/0x8400/0x8400/0.BE.J.
```

Checked voltage on boot switch pin = 1.68V.  

Any suggestion? 

Could it be weak pull up? maybe add stronger external pull up?
Or it noisy external power? Maybe add extra ceramic capacitor?

orangecms, 2024-05-11:
Hey there,

I'm reversing the mask ROMs at this point, have bare metal code running on all the Duo boards - Duo, Duo S and Duo 256M. Curious to see if I can reproduce your issue in a simpler environment. I can tell that there are watchdog timers set up etc.. Could you tell me what the rough timeout is? Is it some minutes or less?

I will publish code at some point that I have running, both on the RISC-V and the Arm core.
Some logs here: https://gist.github.com/orangecms/6df06421cace3d8d1644de583983b826

orangecms, 2024-05-12:
see also #68

hellvesper, 2024-05-14:
> Hey there,
>
> I'm reversing the mask ROMs at this point, have bare metal code running on all the Duo boards - Duo, Duo S and Duo 256M. Curious to see if I can reproduce your issue in a simpler [...]

Hi, sorry for long reply. Well as I said it can work hours in idle. But few seconds with active camera and NPU. May be there is issue with some watchdog timers, I don't know. But why it reboots with ARM core is mystery. And I think, by some reason something trigger boot switch pin which controls which core will be active and this the issue why the hardware reset itself. But again what the source of this issue, I don't know. 

I'm not familiar with chip architecture yet and not sure how cores share gpio control, just suggestion, maybe other core interfere with this pin, I didn't check this part of hardware yet. Could be a hardware issue with chip revision,  which is not in Errate yet also.