milkv-duo/duo-buildroot-sdk #94: How to build host-tools

Binary141, 2024-05-15:
I noticed when downloading the `host-tools` using the `build.sh` script that there are the `riscv64-elf-x86_64`, `riscv64-linux-musl-x86_64`, and  `riscv64-linux-x86_64` folders. This leads me to believe I can build from the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) repo to get the same files and executables. 

Building using the default `./configure --prefix=/opt/riscv --enable-multilib` or `./configure --prefix=/opt/riscv-elf --enable-multilib --with-abi=lp64d --with-arch=rv64gc` I get some errors while building for the milkv-duo board. I get one error of `error: conflicting types for 'cmd_process' due to enum/integer mismatch;` and another for `Error: unrecognized opcode `csrr a5,0xc01', extension `zicsr' required`.

This leads me to believe I am not building with the correct options or using a different version (built using gcc v13 whereas host-tools seems to have a v10). Are there steps to be able to build the host-tools folder from scratch, or is there a config that could be shared to be able to reproduce the executables and libraries?

karaketir16, 2025-02-23:
I got the same issue
`error: conflicting types for 'cmd_process' due to enum/integer mismatch;`

Have you solved it?

Binary141, 2025-02-23:
I eventually ended up compiling from the [riscv-gnu-toolchail](https://github.com/riscv-collab/riscv-gnu-toolchain) and took a while to get the options correct. I think in the end I ended up with this configuration set to get the code to compile and working. Note I compiled this so long ago that I could be wrong, and I did end up getting assembly (and C) to be compiled and uploaded straight to the Duo without the use of the toolchain and playing around with it [here](https://github.com/Binary141/milkv-duo-playground)

```
    --with-system-zlib \
    --disable-multilib \
    --with-abi=lp64d \
    --with-arch=rv64imafdc \
    --with-tune=rocket \
    --with-isa-spec=20191213 \
    'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' \
    'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
```

karaketir16, 2025-02-23:
Your repository looks great, thanks for sharing. Is there any documentation/explanation that I can follow and learn? Or do you know any similar projects/documents?

I will try to boot with your repository.

I will probably try it with https://github.com/XUANTIE-RV/xuantie-gnu-toolchain (with prebuild version)