milkv-duo/duo-buildroot-sdk #95: Duo 256 MB - camera-test.sh kernel panic under low light conditions

teknoman117, 2024-05-15:
Related to #70 

I noticed a very strange pattern (to me) with kernel panics and the camera on the Duo 256 MB. I tried the camera sample again this morning and it was miraculously working! I was playing with the RTSP stream for a short while, but the moment I set the camera down the Duo kernel panicked. I thought that it might be due to a poor connection to the camera so I reconnected everything. I started camera-test.sh while holding the camera again and it was working again. Wiggled the cables, etc. Seemed fine.

I then waved my hand in front of the camera and boom - kernel panic. I noticed that if I start the same with the camera in a well lit room, and then slowly block the light coming into the camera, it will panic when the scene starts to get dark.

I modified the *sample_vi_fd* example to remove both the NPU usage and the RTSP stream - essentially just capturing frames from the camera. The crash/panic still happens even when the video encoder and NPU are not being fed data. They always come from random parts of the kernel and from garbage instruction pointers, which would suggest to me that whatever is going wrong is corrupting kernel memory.

Could this be some kind of ISP bug related to auto-exposure / auto-focus? Are there any userspace or TDL SDK options I can tweak to reduce this behavior?

carbonfix, 2024-05-29:
Please try using firmware V1.1.1 or later.
https://github.com/milkv-duo/duo-buildroot-sdk/releases

teknoman117, 2024-05-29:
I can confirm that it is indeed working now. Thank you!

I see two commits that seem to reference the Duo 256M, are either/both of these the fix for this?

https://github.com/milkv-duo/duo-buildroot-sdk/commit/e49d4243dd0d843ce7f7d56522fdee52a0eff180
https://github.com/milkv-duo/duo-buildroot-sdk/commit/540dc3e57f93253b1920a952c9404ae247f22480

What do these do? What did TPU_PERF_MODE do?

closed 2024-05-29 by teknoman117

teknoman117, 2024-05-29:
Question was answered on the forums - reduced the TPU from 1 TOP/s to 0.5 TOP/s.