milkv-duo/duo-buildroot-sdk #97: Duo S SD image not building

Plaque-fcc, 2024-05-23:
Hi, with `./build.sh milkv-duos-sd` I get to this point:

```
/buildbot/milkv/duo-buildroot-sdk_new00/device/gen_burn_image_sd.sh /buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd
BR_DIR: /buildbot/milkv/duo-buildroot-sdk_new00/buildroot-2021.05
BR_BOARD: milkv-duos-sd_musl_riscv64
/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd
~/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd ~/milkv/duo-buildroot-sdk_new00/build
/buildbot/milkv/duo-buildroot-sdk_new00/device/milkv-duos-sd/genimage.cfg:39: no sub-section title/index for 'config'
INFO: cmd: "mkdir -p "/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/tmp"" (stderr):
INFO: cmd: "rm -rf "/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/tmp"" (stderr):
INFO: cmd: "cp -a "/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/fs/" "/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/tmp/root"" (stderr):
INFO: cmd: "find '/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/fs//{}' '/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/tmp/root/{}'" (stderr):
ERROR: file(logo.jpg): stat(/buildbot/milkv/duo-buildroot-sdk_new00/install/soc_cv1813h_milkv_duos_sd/logo.jpg) failed: No such file or directory
ERROR: hdimage(milkv-duos-sd.img): could not setup logo.jpg
gnimage for milkv-duos-sd failed!
~/milkv/duo-buildroot-sdk_new00/build
~/milkv/duo-buildroot-sdk_new00
Error: Create SD image failed!
```

It looks very much as if there's some code missing around these recent changes about disabling-enabling logo.

Commit 840b72db19a5ad293d2c22892e23fccdcc573542 (current **develop** branch).

abehnke, 2024-05-23:
I received this same error error earlier today using the same command ./build.sh milkv-duos-sd .   In past days, I was able to build images for the smaller milkv duo with no problem.

Plaque-fcc, 2024-05-23:
5d8e68032 (duos disable bootlogo by default) is where it stops building.

Sfinx, 2024-05-24:
> [5d8e680](https://github.com/milkv-duo/duo-buildroot-sdk/commit/5d8e6803215c746db6986b260c99b547adc7ec55) (duos disable bootlogo by default) is where it stops building.

it still fails with this commit

abehnke, 2024-05-24:
I edited my local copy of the SDK by removing the logo code.  I don't really know what I am doing.  I just guessed that removing code referencing the logo might work.  The image now builds for me.

```
diff --git a/build/boards/cv181x/cv1813h_milkv_duos_sd/partition/partition_sd.xml b/build/boards/cv181x/cv1813h_milkv_duos_sd/partition/partition_sd.xml
index 2cfc1e382..1fbea7300 100644
--- a/build/boards/cv181x/cv1813h_milkv_duos_sd/partition/partition_sd.xml
+++ b/build/boards/cv181x/cv1813h_milkv_duos_sd/partition/partition_sd.xml
@@ -1,5 +1,4 @@
 <physical_partition type="sd">
     <partition label="BOOT" size_in_kb="131072" readonly="false" file="boot.sd"/>
-    <partition label="MISC" size_in_kb="2048" readonly="false" file="logo.jpg"/>
     <partition label="ROOTFS" size_in_kb="768432" readonly="false" file="rootfs.sd" />
 </physical_partition>
diff --git a/build/boards/cv181x/cv1813h_milkv_duos_sd/u-boot/cvitek_cv1813h_milkv_duos_sd_defconfig b/build/boards/cv181x/cv1813h_milkv_duos_sd/u-boot/cvitek_cv1813h_milkv_duos_sd_defconfig
index 13d629625..d9b2ff436 100644
--- a/build/boards/cv181x/cv1813h_milkv_duos_sd/u-boot/cvitek_cv1813h_milkv_duos_sd_defconfig
+++ b/build/boards/cv181x/cv1813h_milkv_duos_sd/u-boot/cvitek_cv1813h_milkv_duos_sd_defconfig
@@ -42,7 +42,7 @@ CONFIG_LZMA=y
 # CONFIG_TOOLS_LIBCRYPTO is not set
 CONFIG_ENV_IS_NOWHERE=y
 
-CONFIG_BOOTLOGO=y
+# CONFIG_BOOTLOGO is not set
 CONFIG_CMD_CVI_VO=y
 CONFIG_CMD_CVI_JPEG=y
 CONFIG_CMD_GPIO=y
diff --git a/device/milkv-duos-sd/genimage.cfg b/device/milkv-duos-sd/genimage.cfg
index 06c801547..d3f55e37d 100644
--- a/device/milkv-duos-sd/genimage.cfg
+++ b/device/milkv-duos-sd/genimage.cfg
@@ -26,11 +26,6 @@ image milkv-duos-sd.img {
                 image = "boot.vfat"
 	}
 
-	partition logo {
-                image = "logo.jpg"
-		size = 2M
-	}
-
 	partition rootfs {
 		partition-type = 0x83
 		image = "rootfs.ext4"

```

Sfinx, 2024-05-24:
Manually copying the device/milkv-duos-sd/logo.jpg to install/soc_cv1813h_milkv_duos_sd/logo.jpg and issuing 'pack_sd_image' works too

Plaque-fcc, 2024-05-25:
> > [5d8e680](https://github.com/milkv-duo/duo-buildroot-sdk/commit/5d8e6803215c746db6986b260c99b547adc7ec55) (duos disable bootlogo by default) is where it stops building.
> 
> it still fails with this commit

I did `git checkout develop && git checkout -b develop-logo && git revert 5d8e6803215c746db6986b260c99b547adc7ec55` — and it works for now.
The way with copying logo file manually by @Sfinx will produce the same thing as this one.
The one by @abehnke implies you don't need the logo, I suppose.

Plaque-fcc, 2024-05-26:
Works after d41e90e48af3c649636a74554d640405b289b677. Thanks.

closed 2024-05-26 by Plaque-fcc