milkv-duo/duo-buildroot-sdk #98: The RTSP video streaming from camera-test.sh doesn't work

cmirceac, 2024-06-01:
Since version 1.1.1 the **camera-test.sh** script starts and works normally (?), but the video streaming doesn't seem to work on either VLC or Celluloid player on latest Linux Mint, all updates are installed, no custom stuff, the codecs are installed and other streams are playing OK.

On the camera side, the script produces the following messages on repeated attempts to connect with different players:

`RTSP client connected from: 10.42.0.1
RTSP client disconnected from: 10.42.0.1
RTSP client connected from: 10.42.0.1
RTSP client connected from: 10.42.0.1
sendRTPorRTCPPacketOverTCP: failed! (errno 104)
RTSP client disconnected from: 10.42.0.1
face count: 1
face count: 0`

Also on the system log there are a TON of rather cryptic messages, but it seems seems that is not happy:

`
[  943.055805] Pool[0] vb paddr(0x8b5f7600) usr_cnt(1) /USER/
[  943.055817] Pool[0] vb paddr(0x8b8eec00) usr_cnt(1) /VPSS/
[  943.055830] VB_POOL owner(0xffffffff) poolID(0x1) pool is empty.
[  943.055842] Pool[1] vb paddr(0x8bbe7000) usr_cnt(1) /VPSS/
[  943.055854] Pool[1] vb paddr(0x8bede600) usr_cnt(1) /VPSS/
[  943.055866] Pool[1] vb paddr(0x8c1d5c00) usr_cnt(1) /USER/
[  943.087807] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17224
[  943.087855] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  943.087948] cvi_img_device_run:437(): ****img(1) grp(0) start****
[  943.087966] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  943.121121] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17225
[  943.123804] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  943.153861] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  943.153880] cvi_vip_online_check_sc_rdy:623(): No sc-1 buffer available.
[  944.055321] vpss_handle_online_frame_done:2439(): grp(0) chn(0) end
[  944.055342] pool(1) vb(0x8bede600) release, Try acquire vb for VPSS
[  944.055363] vpss_handle_online_frame_done:2439(): grp(0) chn(1) end
[  944.055399] VB_POOL owner(0xffffffff) poolID(0x0) pool is empty.
[  944.055413] Pool[0] vb paddr(0x8b300000) usr_cnt(1) /VPSS/
[  944.055425] Pool[0] vb paddr(0x8b5f7600) usr_cnt(1) /USER/
[  944.055437] Pool[0] vb paddr(0x8b8eec00) usr_cnt(1) /VPSS/
[  944.055450] VB_POOL owner(0xffffffff) poolID(0x1) pool is empty.
[  944.055506] Pool[1] vb paddr(0x8bbe7000) usr_cnt(1) /VPSS/
[  944.055519] Pool[1] vb paddr(0x8bede600) usr_cnt(1) /VPSS/
[  944.055531] Pool[1] vb paddr(0x8c1d5c00) usr_cnt(1) /USER/
[  944.087483] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17254
[  944.087554] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  944.087649] cvi_img_device_run:437(): ****img(1) grp(0) start****
[  944.087666] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  944.120797] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17255
[  944.123480] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  944.153529] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  944.153547] cvi_vip_online_check_sc_rdy:623(): No sc-1 buffer available.
[  945.055054] vpss_handle_online_frame_done:2439(): grp(0) chn(1) end
[  945.055089] VB_POOL owner(0xffffffff) poolID(0x0) pool is empty.
[  945.055104] Pool[0] vb paddr(0x8b300000) usr_cnt(1) /VPSS/
[  945.055116] Pool[0] vb paddr(0x8b5f7600) usr_cnt(1) /USER/
[  945.055128] Pool[0] vb paddr(0x8b8eec00) usr_cnt(1) /VPSS/
[  945.055141] VB_POOL owner(0xffffffff) poolID(0x1) pool is empty.
[  945.055153] Pool[1] vb paddr(0x8bbe7000) usr_cnt(1) /VPSS/
[  945.055164] Pool[1] vb paddr(0x8bede600) usr_cnt(1) /VPSS/
[  945.055176] Pool[1] vb paddr(0x8c1d5c00) usr_cnt(1) /USER/
[  945.087158] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17284
[  945.087205] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  945.087299] cvi_img_device_run:437(): ****img(1) grp(0) start****
[  945.087317] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  945.120472] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17285
[  945.123152] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  945.153206] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  945.153223] cvi_vip_online_check_sc_rdy:623(): No sc-1 buffer available.
[  945.153797] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17286
[  945.153843] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  946.054746] VB_POOL owner(0xffffffff) poolID(0x0) pool is empty.
[  946.054761] Pool[0] vb paddr(0x8b300000) usr_cnt(1) /VPSS/
[  946.054773] Pool[0] vb paddr(0x8b5f7600) usr_cnt(1) /USER/
[  946.054785] Pool[0] vb paddr(0x8b8eec00) usr_cnt(1) /VPSS/
[  946.054798] VB_POOL owner(0xffffffff) poolID(0x1) pool is empty.
[  946.054810] Pool[1] vb paddr(0x8bbe7000) usr_cnt(1) /VPSS/
[  946.054822] Pool[1] vb paddr(0x8bede600) usr_cnt(1) /VPSS/
[  946.054834] Pool[1] vb paddr(0x8c1d5c00) usr_cnt(1) /USER/
[  946.086832] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17314
[  946.086879] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  946.086973] cvi_img_device_run:437(): ****img(1) grp(0) start****
[  946.086990] isp_pre_trig:451(): trigger fe_0 chn_num_0 frame_vld
[  946.120146] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17315
[  946.122828] _isp_postraw_shaw_done_handler:7126(): postraw shaw done
[  946.152882] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)
[  946.152900] cvi_vip_online_check_sc_rdy:623(): No sc-1 buffer available.
[  946.153473] vi_irq_handler:7325(): pre_fe_0 sof chn_num=0 frm_num=17316
[  946.153521] cvi_sc_cfg_cb:875(): online trig for snr-0. img(1:0:3)^C
`

the players just report that they can't play the stream :(.

Please help me find the issue or suggest another working way to be able to stream the video properly.

Many thanks,

teknoman117, 2024-06-02:
@cmirceac I have ffmpeg 6.1.1 installed on my system and I am able to view the video stream with

```
ffplay rtsp://192.168.42.1/h264
```

cmirceac, 2024-06-02:
I can confirm, with ffplay from ffmpeg-4.4.2 from official Linux Mint it works as well, it even draws those green rectangles when human faces are being recognized :).
But I think I have found the issue with more "mainstream" players, like VLC are not working and giving up, immediately when the ffplayer is started a number of frames are coming invalid and that's probably what makes the other players abort the playing. Interesting is that it happens  only at the beginning, afterwards, even after hours of playing there are no more errors.
I'll leave the bug open as it seems to be a regression, the original Wiki documentation shows it playing with VLC.
Maybe the script needs a bit of tweaking to avoid those bad frames at the beginning.

These are the error messages that show at startup of streaming:

$ ffplay  rtsp://10.42.0.76/h264
ffplay version 4.4.2-0ubuntu0.22.04.1 Copyright (c) 2003-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-19ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  WARNING: library configuration mismatch
  avcodec     configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --enable-version3 --disable-doc --disable-programs --enable-libaribb24 --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libtesseract --enable-libvo_amrwbenc --enable-libsmbclient
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[h264 @ 0x7f3d98006240] non-existing PPS 0 referenced    0B f=0/0   
    Last message repeated 1 times
[h264 @ 0x7f3d98006240] decode_slice_header error
[h264 @ 0x7f3d98006240] no frame!
[h264 @ 0x7f3d98006240] non-existing PPS 0 referenced    0B f=0/0   
    Last message repeated 1 times
[h264 @ 0x7f3d98006240] decode_slice_header error
[h264 @ 0x7f3d98006240] no frame!
[h264 @ 0x7f3d98006240] non-existing PPS 0 referenced    0B f=0/0   
    Last message repeated 1 times
[h264 @ 0x7f3d98006240] decode_slice_header error
[h264 @ 0x7f3d98006240] no frame!
Input #0, rtsp, from 'rtsp://10.42.0.76/h264':0KB sq=    0B f=0/0   
  Metadata:
    title           : 
    comment         : h264
  Duration: N/A, start: 0.032678, bitrate: N/A
  Stream #0:0: Video: h264 (Constrained Baseline), yuv420p(progressive), 1920x1080, 30 tbr, 90k tbn, 180k tbc
[rtsp @ 0x7f3d98000cc0] max delay reached. need to consume packet   
[rtsp @ 0x7f3d98000cc0] RTP: missed 30 packets
[h264 @ 0x7f3d980c2280] Invalid level prefix029KB sq=    0B f=0/0   
[h264 @ 0x7f3d980c2280] error while decoding MB 66 54
[h264 @ 0x7f3d980c2280] concealing 1663 DC, 1663 AC, 1663 MV errors in P frame
 591.34 M-V: -0.016 fd=   7 aq=    0KB vq=  551KB sq=    0B f=0/0

teknoman117, 2024-06-02:
It seems that when a new client connects it starts immediately receiving whatever the last packet coming out of the encoder was. So you're going to not be able to decode frames until another I-frame comes along. It looks like the samples repo uses a GOP size of 60 and the video encoder is running at 25 fps. So basically  2.4 seconds between keyframes.

A simple fix would be to have _sample_vi_fd_ (the program camera-test.sh runs) not send any data to a newly connected client until the encoder spits out another reference frame.

cmirceac, 2024-06-02:
It seems that somehow **sample_vi_fd** is included in binary format in the image and is actually part of **cvitek-tdl-sdk-cv180x**, have reported the issue there and I will close the issue here as is not directly related with the SDK.

closed 2024-06-02 by cmirceac