milkv-duo/duo-buildroot-sdk #114: No Arduino Support

TeslaLaserRay, 2024-06-25:
In any newer version than 1.0.9 is no Support for Arduino, please fix that.

mikekgr, 2024-06-25:
I came to github to ask the same question : When gentlemen will have the Arduino functionality back.
Please increase this task priority.

Thanks

BoryaGames, 2024-07-03:
Yes, please do that, I'm already orderer Duo S, and one of the main reasons I need it - for arduino.

BoryaGames, 2024-07-18:
I got it, seems to work, still no arduino :(

mikekgr, 2024-07-31:
@carbonfix  Please give us an answer when we will have Arduino support updated. Thanks !

carbonfix, 2024-08-01:
We will update the version next week. If you need it urgently, you can compile it yourself using the arduino branch, the code has been updated :)

mikekgr, 2024-08-01:
> We will update the version next week. If you need it urgently, you can compile it yourself using the arduino branch, the code has been updated :)

Thanks a lot. I appreciate your work. I think, it's better to wait for you official release.

carbonfix, 2024-08-01:
> > We will update the version next week. If you need it urgently, you can compile it yourself using the arduino branch, the code has been updated :)
> 
> Thanks a lot. I appreciate your work. I think, it's better to wait for you official release.

Already released in advance:
https://github.com/milkv-duo/duo-buildroot-sdk/releases/tag/Duo-V1.1.2

closed 2024-08-01 by carbonfix

BoryaGames, 2024-08-02:
> Already released in advance: https://github.com/milkv-duo/duo-buildroot-sdk/releases/tag/Duo-V1.1.2

Duo S still doesn't support arduino, any ETA for it, please?