milkv-duo/duo-buildroot-sdk #123: Missing support for RPi camera (sensor IMX219)

silentjet, 2024-07-20:
RPi camera module of the first version is rather rarely available these days, instead the most common is RPi camera module v2, which uses a different sensor - IMX219. Please add the support for this camera module....

Getyorma, 2024-07-23:
I just today hooked it up expecting it to work right away, but: `init middleware failed! ret=ffffffff`. 
The cameras should have identical pin-outs. It seems some drivers are already out there: [IMX219 driver for linux(github.com)](https://github.com/torvalds/linux/blob/master/drivers/media/i2c/imx219.c)

Getyorma, 2024-07-25:
After a bit of digging i found out there are traces of it in Milk-v buildroot: [https://github.com/milkv-duo/duo-buildroot-sdk/blob/99f93483206a0691a3b5d1afe0aa5db4816c4e6d/linux_5.10/drivers/media/i2c/imx219.c#L121](https://github.com/milkv-duo/duo-buildroot-sdk/blob/99f93483206a0691a3b5d1afe0aa5db4816c4e6d/linux_5.10/drivers/media/i2c/imx219.c#L121)

I am not too good at Milk-V embedded Linux OS but maybe we only need to add an appropriate configuration "sensor_cfg.ini" to /mnt/data/sensor_cfg.ini in order to make it work... I will tinker with this later and report if i there is more progress after creating the correct .ini... I do not know if the lane_id is the same or not, but i would not know where to look to find out... I do not know if i need to pinmux stuff to make the camera work etc. I know that in order to use I2C one has to pinmux the SDA,SCL pins. Disregarding all worries, the github link above suggests that it should be pretty much plug and play.

silentjet, 2024-07-25:
So when we talk about different RPI cameras on a physical level (by pin designation) they are the same or compatible. What IS different is a sensor's bus configuration, because the configuration depends on the sensor hardware. Normally it shall be configured via device tree files (remember that Rpi's *.dtbo files?), but here is taken a different approach...

piotr-defendeye, 2024-08-21:
Has anyone had any success with this?

raphaelscholle, 2024-08-30:
getting new cameras working is really hard and expensive, I did glance at this repo a while, but didn't really work on anything here.
Usually I'll just write a device tree overlay and make the driver work, (everything else is usually commissioned).But it looks like this is a little different.

raphaelscholle, 2024-08-30:
you also forget something important .. the rpi-mipi-csi port is 3.3V logic level, the civitec one is 1.8, so a adapter should always be needed