milkv-duo/duo-buildroot-sdk #133: Failed to build Milk-V Duo Image

ArielHeleneto, 2024-08-29:
# Build with build.sh

```
buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/libgcc.a(trunctfdf2.o): mis-matched ISA version 0.7 for 'zvamo' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/libgcc.a(trunctfdf2.o): mis-matched ISA version 0.7 for 'zvlsseg' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/libgcc.a(_clzsi2.o): mis-matched ISA version 0.7 for 'v' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/libgcc.a(_clzsi2.o): mis-matched ISA version 0.7 for 'zvamo' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/libgcc.a(_clzsi2.o): mis-matched ISA version 0.7 for 'zvlsseg' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/crtend.o: mis-matched ISA version 0.7 for 'v' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/crtend.o: mis-matched ISA version 0.7 for 'zvamo' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/crtend.o: mis-matched ISA version 0.7 for 'zvlsseg' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/crtn.o: mis-matched ISA version 0.7 for 'v' extension, the output version is 1.0
/home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: /home/ariel/duo-buildroot-sdk/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/lib64v0p7_xthead/lp64d/crtn.o: mis-matched ISA version 0.7 for 'zvamo' extension, the output version is 1.0
```

ArielHeleneto, 2024-08-29:
使用 Docker 和直接构建都能观察到类似的问题，最后一段输出如下。

```
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_gmm2
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_xor
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_normgrad
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_ordstatfilter
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_sad
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_vcodec
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_osdc
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_integ
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_ncc
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_and
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_dilate
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_stcandicorner
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_framediffmotion
striping /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/mnt/system/usr/bin/sample_sobel
mkdir -p /home/work/buildroot-2021.05/board/milkv/milkv-duo-sd/overlay
cp -arf /home/work/install/soc_cv1800b_milkv_duo_sd/tmp-rootfs/* /home/work/buildroot-2021.05/board/milkv/milkv-duo-sd/overlay
# create partition mounting points and move sytems to read-write partitions
  [TARGET] br-rootfs-pack 
make -C /home/work/buildroot-2021.05 milkv-duo-sd_musl_riscv64_defconfig BR2_TOOLCHAIN_EXTERNAL_PATH=/home/work/host-tools/gcc/riscv64-linux-musl-x86_64
make[1]: Entering directory '/home/work/buildroot-2021.05'
#
# configuration written to /home/work/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/.config
#
make[1]: Leaving directory '/home/work/buildroot-2021.05'
/home/work/buildroot-2021.05/utils/brmake -j16 -C /home/work/buildroot-2021.05
2024-08-29T19:13:08 >>> host-util-linux 2.36.2 Building
2024-08-29T19:13:08 >>> host-autoconf 2.69 Configuring
2024-08-29T19:13:08 >>> ncurses 6.1 Building
2024-08-29T19:13:08 >>> libzlib 1.2.11 Installing to staging directory
2024-08-29T19:13:08 >>> expat 2.4.1 Building
2024-08-29T19:13:08 >>> libzlib 1.2.11 Fixing libtool files
Done in 11s  (error code: 2)
make: *** [Makefile:597: br-rootfs-pack] Error 2
Error: Build board milkv-duo-sd failed!
```

屏幕录像见 [bugs](https://github.com/ArielHeleneto/Work-PLCT/tree/master/duo/bug)

ArielHeleneto, 2024-08-29:
问题出在 https://github.com/milkv-duo/duo-buildroot-sdk/commit/2c5bc4c38ef6f67b0213c5e22f16c58e72502225

carbonfix, 2024-08-30:
> 问题出在 [2c5bc4c](https://github.com/milkv-duo/duo-buildroot-sdk/commit/2c5bc4c38ef6f67b0213c5e22f16c58e72502225)

所以现在问题解决了么？

ArielHeleneto, 2024-08-30:
> > 问题出在 [2c5bc4c](https://github.com/milkv-duo/duo-buildroot-sdk/commit/2c5bc4c38ef6f67b0213c5e22f16c58e72502225)
> 
> 所以现在问题解决了么？

无进展

ArielHeleneto, 2024-09-01:
该问题似乎和 #49 类似。

检查 build/br.log 发现下列可疑内容

```
2024-09-01T15:37:32 CC       lib/swapon-swapprober.o                                                                                                                                            
2024-09-01T15:37:32 CC       sys-utils/swapoff-swapoff.o                                                                                                                                        
2024-09-01T15:37:32 CC       sys-utils/swapoff-swapon-common.o                                                                                                                                  
2024-09-01T15:37:32 CC       lib/swapoff-swapprober.o                                                                                                                                           
2024-09-01T15:37:32 misc-utils/kill.c: 在函数‘kill_with_timeout’中:                                                                                                                             
2024-09-01T15:37:32 misc-utils/kill.c:394:20: 错误：implicit declaration of function ‘pidfd_open’; did you mean ‘fdopen’? [-Wimplicit-function-declaration]                                     
2024-09-01T15:37:32 394 |         if ((pfd = pidfd_open(ctl->pid, 0)) < 0)                                                                                                                      
2024-09-01T15:37:32 |                    ^~~~~~~~~~                                                                                                                                             
2024-09-01T15:37:32 |                    fdopen                                                                                                                                                 
2024-09-01T15:37:32 misc-utils/kill.c:394:20: 警告：对‘pidfd_open’的嵌套的外部声明 [-Wnested-externs]                                                                                           
2024-09-01T15:37:32 misc-utils/kill.c:399:13: 错误：implicit declaration of function ‘pidfd_send_signal’; did you mean ‘SYS_pidfd_send_signal’? [-Wimplicit-function-declaration]               
2024-09-01T15:37:32 399 |         if (pidfd_send_signal(pfd, ctl->numsig, &info, 0) < 0)                                                                                                        
2024-09-01T15:37:32 |             ^~~~~~~~~~~~~~~~~                                                                                                                                             
2024-09-01T15:37:32 |             SYS_pidfd_send_signal                                                                                                                                         
2024-09-01T15:37:32 misc-utils/kill.c:399:13: 警告：对‘pidfd_send_signal’的嵌套的外部声明 [-Wnested-externs]                                                                                    
2024-09-01T15:37:32 make[4]: *** [Makefile:9008：misc-utils/kill.o] 错误 1                                                                                                                      
2024-09-01T15:37:32 make[4]: *** 正在等待未完成的任务....                                                                                                                                       
2024-09-01T15:37:32 /usr/bin/ar: `u' modifier ignored since `D' is the default (see `U')                                                                                                        
2024-09-01T15:37:32 /usr/bin/ar: `u' modifier ignored since `D' is the default (see `U')                                                                                                        
2024-09-01T15:37:33 make[4]: 离开目录“/home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/build/host-util-linux-2.36.2”                                            
2024-09-01T15:37:33 make[3]: *** [Makefile:13799：all-recursive] 错误 1                                                                                                                         
2024-09-01T15:37:33 make[3]: 离开目录“/home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/build/host-util-linux-2.36.2”                                            
2024-09-01T15:37:33 make[2]: *** [Makefile:5899：all] 错误 2                                                                                                                                    
2024-09-01T15:37:33 make[2]: 离开目录“/home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/build/host-util-linux-2.36.2”                                            
2024-09-01T15:37:33 make[1]: *** [package/pkg-generic.mk:250：/home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/build/host-util-linux-2.36.2/.stamp_built] 错误 22024-09-01T15:37:33 make[1]: *** 正在等待未完成的任务....                                                                                                                                       
2024-09-01T15:37:33 >>> host-automake 1.15.1 Configuring                                                                                                                                        
2024-09-01T15:37:33 mkdir -p /home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/per-package/host-automake/host                                                    
[... cut ...]
2024-09-01T15:37:33 rsync -a --link-dest=/home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/per-package/host-skeleton/host/ /home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/per-package/host-skeleton/host/ /home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/per-package/host-automake/host      
2024-09-01T15:37:33 mkdir -p /home/ariel/duo-buildroot-sdk/buildroot-2021.05/output/milkv-duo-sd_musl_riscv64/per-package/host-automake/target
```