milkv-duo/duo-buildroot-sdk #143: Question: Autostart camera

PaulaBras, 2024-09-24:
Heyho,
I want to autostart the camera, but I am not really able to.
I tried this:
```
[root@milkv-duo]~# cat /mnt/system/auto.sh
#!/bin/sh

# Put the program you want to run automatically here
#/mnt/system/blink.sh

export LD_LIBRARY_PATH=/mnt/system/lib

sleep 15 && sample_vi_fd /mnt/cvimodel/scrfd_768_432_int8_1x.cvimodel > /dev/null 2>&1 &
```

But it didn't work. When I run the command manually after the startup it will run in the background with no output etc. also after closing the shell session. Is there any more reliable way to autostart the camera?

Thanks in advance.

carbonfix, 2024-09-25:
Please try using the absolute path for sample_vi_fd:
```
sleep 15 && /mnt/system/usr/bin/sample_vi_fd /mnt/cvimodel/scrfd_768_432_int8_1x.cvimodel > /dev/null 2>&1 &
```

PaulaBras, 2024-09-25:
thanks that worked!

closed 2024-09-25 by PaulaBras

alpha027, 2025-04-20:
In the new version of the Duo image (Nov 2024 Release), the following line should be commented out:

`
export LD_LIBRARY_PATH=/mnt/system/lib`

The code is simply:

```
[root@milkv-duo]~# cat /mnt/system/auto.sh
#!/bin/sh

# Put the program you want to run automatically here
#/mnt/system/blink.sh

sleep 15 && sample_vi_fd /mnt/cvimodel/scrfd_768_432_int8_1x.cvimodel > /dev/null 2>&1 &
```

alpha027, 2025-04-20:
Unfortunately this response is no longer working after the release of the new image (V1.1.4 of November 2024). I have tested the answer above and it worked in the previous release. In my case, I am using the MilkV Duo S board.

In the new release, I am testing this using the `sample_vi_od` executable that runs like this: 
`/root/sample_vi_od yolov8-person-pets /root/pet_det_640x384.cvimodel`

First, I found that having the `export LD_LIBRARY_PATH=/mnt/system/lib` line gave errors:
```
Error loading shared library libcvi_ive.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libini.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libsns_full.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libsample.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libisp.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libvdec.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libvenc.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libawb.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
Error loading shared library libae.so: No such file or directory (needed by /mnt/system/usr/bin/sample_vi_od)
...
```
Second, removing the `export LD_LIBRARY_PATH=/mnt/system/lib` line fixed the execution of the code, the rtsp feed could be viewed on VLC player. **However, the code still didn't autostart.** I can only manually execute auto.sh, it works, but I don't get any streamed feed when replugging the board.

PaulaBras, 2025-04-21:
Hi, sadly I cannot reproduce the issue you got.
I also used the current latest (v1.1.4) image and I still need to add the LD_LIBARY_PATH to get a video feed. I also encountered the non autostart issue. It was a mix of the missing LD_LIBRARY_PATH and a date/time issue.

My auto.sh looks like this:
```bash
#!/bin/sh

# Redirect all output to a log file for debugging
exec > /mnt/system/camera_autostart.log 2>&1

# Log the start time
date

# Check if blink.sh exists
if [ -f /mnt/system/blink.sh ]; then
  mv /mnt/system/blink.sh /mnt/system/blink.bak
  echo "blink.sh moved to blink.bak"
fi

# Disable LED
LED_PIN=509
LED_GPIO=/sys/class/gpio/gpio${LED_PIN}
if test -d ${LED_GPIO}; then
  echo "PIN ${LED_PIN} already exported"
else
  echo ${LED_PIN} > /sys/class/gpio/export
fi
echo out > ${LED_GPIO}/direction
echo 0 > ${LED_GPIO}/value

# Increase the delay before starting the camera
echo "Waiting 15 seconds before starting camera..."
sleep 15

# Start Video feed with full path and error checking
echo "Starting camera..."
export LD_LIBRARY_PATH=/mnt/system/lib
/mnt/system/usr/bin/sample_vi_simple_camera
echo "Camera start exit code: $?"
```

I recompiled the camera to also remove the face detection from it. I uploaded it here: [https://git.pabr.de/paul/cvitek-tdl-sdk-cv180x/-/tree/main/sample/cvi_tdl?ref_type=heads](https://git.pabr.de/paul/cvitek-tdl-sdk-cv180x/-/tree/main/sample/cvi_tdl?ref_type=heads) There are 2 versions of it [sample_vi_clean_video.c](https://git.pabr.de/paul/cvitek-tdl-sdk-cv180x/-/blob/main/sample/cvi_tdl/sample_vi_clean_video.c?ref_type=heads) which uses the original code, but the removed face print (detection still happening) and this one [sample_vi_simple_camera.c](https://git.pabr.de/paul/cvitek-tdl-sdk-cv180x/-/blob/main/sample/cvi_tdl/sample_vi_simple_camera.c?ref_type=heads) which I am using since it is only the camera feed. If you want to use the face detection you need to change the 2nd last line and add the parameter back to it. (sample_vi_fd /mnt/cvimodel/scrfd_768_432_int8_1x.cvimodel)

Keep in mind that you still need to compile it yourself and upload it to the corresponding folder and folder via sftp/ssh and make it executable for the root user (chmod u+x <path>)