milkv-duo/duo-buildroot-sdk #144: Add bluetooth support

lamaalrajih, 2024-09-26:
Hi -- we'd like to use the buildroot sdk but it unfortunately doesn't have any bluetooth support. Can you guys please add that?

carbonfix, 2024-09-27:
test in duos:
hciattach -s 1500000 /dev/ttyS4 any 1500000 flow nosleep
hciconfig hci0 up
hcitool scan

BoryaGames, 2025-01-08:
Hello, @carbonfix 
Since that user didn't respond for a long time, I tested it myself.
```
[root@milkv-duo]~# hciattach -s 1500000 /dev/ttyS4 any 1500000 flow nosleep
Device setup complete
[root@milkv-duo]~# hciconfig hci0 up
[root@milkv-duo]~# hcitool scan
Scanning ...
[root@milkv-duo]~# hcitool scan
Scanning ...
[root@milkv-duo]~# hcitool scan
Scanning ...
        20:17:11:08:06:18       U7 Max
```
I can say it found another Bluetooth device successfuly. Can you, please, add more information about this in the docs? Add these commands and also add commands how to connect/disconnect and use it, not only do a scan.