milkv-duo/duo-buildroot-sdk #147: Unable to boot in ARM mode with the provided images

bwees, 2024-10-09:
Hello,

I just got my Duo S and flashed the SD card with https://github.com/milkv-duo/duo-buildroot-sdk/releases/download/v1.1.3/milkv-duos-sd-v1.1.3-2024-0930.img.zip. The board boots fine in RISC-V mode but fails to boot in ARM mode. Is there another set of images to use ARM mode or do I need to compile the image myself?

BoryaGames, 2024-10-10:
Yeah, you need a different image for ARM, of course. And current SDK can't even do that right now, you need to wait for [V2 version](https://github.com/milkv-duo/duo-buildroot-sdk-v2) and then compile it.

bwees, 2024-10-10:
Ah gotcha, yeah after more digging I found that out. Thanks fo the info!

closed 2024-10-10 by bwees