milkv-duo/duo-buildroot-sdk #148: There is no Swap on Duo

nopnop2002, 2024-10-10:
I used v1.1.3 on Duo from here.
https://github.com/milkv-duo/duo-buildroot-sdk/releases

They say:
Duo deleted the third partition reserved for swap and used swap file instead.

However, there is no Swap.

```
# free -h
              total        used        free      shared  buff/cache   available
Mem:          28.3M       13.6M        5.8M       68.0K        8.9M       12.0M
Swap:             0           0           0
```

carbonfix, 2024-10-10:
https://milkv.io/docs/duo/getting-started/setup#enable-swap

nopnop2002, 2024-10-10:
@carbonfix 
Thank you fo replay.

```
[root@milkv-duo]~# fallocate -l 256M /mnt/swapfile
chmod 600 /mnt/swapfile
[root@milkv-duo]~# chmod 600 /mnt/swapfile
mkswap /mnt/swapfile
[root@milkv-duo]~# mkswap /mnt/swapfile
swapon /mnt/swapfileSetting up swapspace version 1, size = 256 MiB (268431360 bytes)
no label, UUID=71fc6ff8-0506-4f4a-a471-7d57b9bd6ea7
[root@milkv-duo]~# swapon /mnt/swapfile
[root@milkv-duo]~# free -h
              total        used        free      shared  buff/cache   available
Mem:          28.3M       13.7M        4.3M       68.0K       10.3M       11.9M
Swap:        255.8M           0      255.8M
```

Swap is now enabled.

nopnop2002, 2024-10-10:
Swap is now enabled, but it disappears when I reboot.
Is there a way to enable it permanently?

carbonfix, 2024-10-10:
The method is mentioned at the end of the document link above :)

nopnop2002, 2024-10-10:
@carbonfix 
```
echo "/mnt/swapfile swap swap defaults 0 0" >> /etc/fstab && sync
```

Good!!

closed 2024-10-10 by nopnop2002