milkv-duo/duo-buildroot-sdk #151: *** [Makefile:597: br-rootfs-pack] Error 2 while building milkv-duos-sd-img

u592003sd, 2024-10-15:
~~~
make[1]: Leaving directory '/home/sarvadnya/duo-buildroot-sdk/buildroot-2021.05'
/home/sarvadnya/duo-buildroot-sdk/buildroot-2021.05/utils/brmake -j12 -C /home/sarvadnya/duo-buildroot-sdk/buildroot-2021.05
2024-10-16T00:02:50 >>> host-util-linux 2.36.2 Building
2024-10-16T00:02:50 >>> host-m4 1.4.18 Installing to host directory
2024-10-16T00:02:50 >>> ncurses 6.1 Configuring
2024-10-16T00:02:50 >>> bash-completion 2.11 Installing to target
2024-10-16T00:02:50 >>> expat 2.4.1 Installing to target
2024-10-16T00:02:50 >>> pcre 8.44 Installing to target
2024-10-16T00:02:50 >>> libzlib 1.2.11 Installing to target
2024-10-16T00:02:50 >>> libical 1.0.1 Installing to target
2024-10-16T00:02:50 >>> host-dbus 1.12.20 Configuring
2024-10-16T00:02:50 >>> gzip 1.10 Installing to target
Done in 17s  (error code: 2)
make: *** [Makefile:597: br-rootfs-pack] Error 2
Error: Build board milkv-duos-sd failed!
~~~

I get the following error when I run the command './build.sh milkv-duos-sd' in the buildroot repo...
Using Ubuntu 22.04 on an intel i7-10750H cpu machine.

Please help

DAMEK86, 2024-10-28:
same on my side

Mr-Pine-apple, 2024-10-29:
same on my side

learnerxuyang, 2025-01-21:
Re-download the latest host tools

yangshining, 2025-01-21:
Re-download the latest host tools

u592003sd, 2025-01-23:
Yes this was helpful... manually unziping the latest host-tools and following the 2nd method "Step-by-step Compilation" in ". Compiled using Ubuntu 22.04" section works properly. Thank you for your time.

yunsss9927, 2025-03-24:
same on my side