milkv-duo/duo-buildroot-sdk #158: arduino 分支下，milkv_duo_sd 修改 ION_SIZE 后无法启动

zdtyuiop4444, 2024-10-30:
环境：官方SDK

之前尝试在 milkv_duo_sd 上测试 sample_audio_rtos 时发现 ION 分配失败，排查后发现 arduino 分支下 milkv_duo_sd 默认 ION_SIZE 为 0，尝试修改 ION_SIZE 为 4m，之后烧录到板子上时直接无法进入 linux 系统。

串口输出如下：
```
..4:08:27:210 [    0.241199] Ion: ion_parse_dt_heap_common: id 0 type 2 name carveout align 1000

..4:08:27:211 [    0.249094] Ion: rmem_ion_device_init: heap carveout base 0x0000000083a28700 size 0x0000000000400000 dev (____ptrval____)

..4:08:27:211 [    0.260383] ion_carveout_heap_create, size=0x400000

..4:08:27:236 [    0.265604] cvi_get_rtos_ion_size, rtos ion_size get:0x0

..4:08:27:237 [    0.290569] platform carveout: [ion] add heap id 0, type 2, base 0x83a28700, size 0x400000

..4:08:27:237 [    0.299581] Advanced Linux Sound Architecture Driver Initialized.

..4:08:27:266 [    0.305948] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000

..4:08:27:288 [    0.314943] Oops [#1]

..4:08:27:289 [    0.317359] Modules linked in:

..4:08:27:289 [    0.320593] CPU: 0 PID: 1 Comm: swapper Not tainted 5.10.4-tag- #1

..4:08:27:289 [    0.327041] epc: ffffffe000282ee6 ra : ffffffe000282e94 sp : ffffffe000a5fc40

..4:08:27:317 [    0.334476]  gp : ffffffe00086a898 tp : ffffffe000a60000 t0 : ffffffe000bb7800

..4:08:27:317 [    0.342002]  t1 : 0000001e00000000 t2 : 00000000000000b0 s0 : 0000000000000000

..4:08:27:317 [    0.349528]  s1 : 0000000000000cc0 a0 : 0000000000000001 a1 : 0000000000000001

..4:08:27:327 [    0.357055]  a2 : 0000000000000000 a3 : 0000000000000cc0 a4 : 000000000000000f

..4:08:27:327 [    0.364582]  a5 : ffffffe003a28410 a6 : 0000000000000cc0 a7 : 0000000000000063

..4:08:27:327 [    0.372108]  s2 : 0000000000000004 s3 : 0000000200000022 s4 : ffffffe00074e238

..4:08:27:327 [    0.379635]  s5 : 0000000000000000 s6 : 0000000000000001 s7 : ffffffe00079ae10

..4:08:27:353 [    0.387161]  s8 : 0000000000000004 s9 : 0000000000000cc0 s10: 0000000000000001

..4:08:27:354 [    0.394688]  s11: 0000000000000010 t3 : 0000000000000000 t4 : 8080808080808080

..4:08:27:375 [    0.402213]  t5 : ffffffe003e69698 t6 : ffffffe000a33251

..4:08:27:376 [    0.407769] status: 0000000200000100 badaddr: 0000000000000000 cause: 000000000000000d

..4:08:27:376 [    0.416010] Call Trace:

..4:08:27:376 [    0.418634] [<ffffffe000282ee6>] pcpu_alloc+0x242/0x3e2

..4:08:27:402 [    0.424086] [<ffffffe00020207a>] rdinit_setup+0x26/0x2a

..4:08:27:403 [    0.429554] [<ffffffe000438380>] loopback_xmit+0xae/0xb2

..4:08:27:403 [    0.435113] [<ffffffe000538e0e>] alloc_netdev_mqs+0x9e/0x286

..4:08:27:403 [    0.441020] [<ffffffe000214024>] loopback_net_init+0x26/0x74

..4:08:27:428 [    0.446928] [<ffffffe00021ae0e>] netdev_init+0x2a/0x4e

..4:08:27:428 [    0.452312] [<ffffffe00052ecde>] ops_init.constprop.0+0xb6/0xd2

..4:08:27:428 [    0.458489] [<ffffffe00052ed74>] register_pernet_operations+0x7a/0xa2

..4:08:27:428 [    0.465208] [<ffffffe00052edfa>] register_pernet_device+0x24/0x56

..4:08:27:438 [    0.471567] [<ffffffe00021ae2e>] netdev_init+0x4a/0x4e

..4:08:27:438 [    0.476944] [<ffffffe00021af48>] net_dev_init+0x116/0x16e

..4:08:27:438 [    0.482598] [<ffffffe000228104>] do_one_initcall+0x58/0xf2

..4:08:27:439 [    0.488324] [<ffffffe00020207a>] rdinit_setup+0x26/0x2a

..4:08:27:439 [    0.493791] [<ffffffe000202b38>] kernel_init_freeable+0x186/0x1a6

..4:08:27:468 [    0.500161] [<ffffffe000616df0>] rest_init+0xba/0xbe

..4:08:27:468 [    0.505348] [<ffffffe000616dfe>] kernel_init+0xa/0xd6

..4:08:27:491 [    0.510633] [<ffffffe000616df0>] rest_init+0xba/0xbe

..4:08:27:491 [    0.515832] [<ffffffe00022906c>] ret_from_syscall_rejected+0x8/0xc

..4:08:27:492 [    0.522335] ---[ end trace 52f94406434d5200 ]---

..4:08:27:492 [    0.527142] note: swapper[1] exited with preempt_count 1

..4:08:27:492 [    0.532697] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

..4:08:27:509 [    0.540650] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

tchamelot, 2025-02-21:
Hello,

I have the same issue, i have been able to boot the kernel by reverting the changes on https://github.com/milkv-duo/duo-buildroot-sdk/blob/7bb22e4bb73be3c1f3fba00fb20a121d725db7ac/linux_5.10/arch/riscv/kernel/setup.c#L93 and https://github.com/milkv-duo/duo-buildroot-sdk/blob/7221d23254b821b1bd418fad75c303f84739b857/linux_5.10/arch/riscv/mm/init.c introduced in 7221d23254b821b1bd418fad75c303f84739b857 

I don't understand why this change was introduced and what it does @carbonfix as the author of this commit do you have any clue.

I am investigating to see if the remoteproc drivers works without this modification

tchamelot, 2025-02-22:
I am using this space as a notebook for anyone looking into the same issue:
Maybe related thread https://lore.kernel.org/lkml/8ed31c81-e6fd-a293-8418-5d0d863c7735@ghiti.fr/t/

As far as I understand, the `early_init_fdt_scan_reserved_mem();` as been moved later in the init process to solve remoteproc issue. However it impacts the ION module

tchamelot, 2025-02-22:
I will be trying those two patches:
https://github.com/torvalds/linux/commit/ef69d2559fe91f23d27a3d6fd640b5641787d22e
https://github.com/torvalds/linux/commit/f1581626071c8e37c58c5e8f0b4126b17172a211

tchamelot, 2025-03-04:
Hello all, great news, I was able to boot the arduino branch and to make the camera work with the following patch:
```diff
diff --git a/build/boards/cv180x/cv1800b_milkv_duo_sd/memmap.py b/build/boards/cv180x/cv1800b_milkv_duo_sd/memmap.py
index 99d640134..00bedd45e 100644
--- a/build/boards/cv180x/cv1800b_milkv_duo_sd/memmap.py
+++ b/build/boards/cv180x/cv1800b_milkv_duo_sd/memmap.py
@@ -40,12 +40,14 @@ class MemoryMap:
     # =================
     # Multimedia buffer. Used by u-boot/kernel/FreeRTOS
     # =================
-    ION_SIZE = 0 * SIZE_1M
+    ION_SIZE = 16 * SIZE_1M
     H26X_BITSTREAM_SIZE = 0 * SIZE_1M
     H26X_ENC_BUFF_SIZE = 0
     ISP_MEM_BASE_SIZE = 0 * SIZE_1M
     BOOTLOGO_SIZE = 0 * SIZE_1M
-    FREERTOS_RESERVED_ION_SIZE = H26X_BITSTREAM_SIZE + H26X_ENC_BUFF_SIZE + ISP_MEM_BASE_SIZE + BOOTLOGO_SIZE
+    FREERTOS_RESERVED_ION_SIZE = (
+        H26X_BITSTREAM_SIZE + H26X_ENC_BUFF_SIZE + ISP_MEM_BASE_SIZE + BOOTLOGO_SIZE
+    )
 
     # ION after FreeRTOS
     ION_ADDR = FREERTOS_ADDR - ION_SIZE
diff --git a/freertos/cvitek/task/main/src/main.c b/freertos/cvitek/task/main/src/main.c
index 4c1d191eb..255bc8e8c 100644
--- a/freertos/cvitek/task/main/src/main.c
+++ b/freertos/cvitek/task/main/src/main.c
@@ -118,6 +118,7 @@ int main(void)
 {
 	pre_system_init();
 	printf("CVIRTOS Build Date:%s  (Time :%s) \n", __DATE__, __TIME__);
+	printf("tchamelot build");
 #ifndef __riscv
 	mmu_enable();
 	printf("enable I/D cache & MMU done\n");
diff --git a/linux_5.10/arch/riscv/kernel/setup.c b/linux_5.10/arch/riscv/kernel/setup.c
index 213cfc8b6..87605d64b 100644
--- a/linux_5.10/arch/riscv/kernel/setup.c
+++ b/linux_5.10/arch/riscv/kernel/setup.c
@@ -90,7 +90,6 @@ void __init setup_arch(char **cmdline_p)
 		pr_err("No DTB found in kernel mappings\n");
 #endif
 
-	early_init_fdt_scan_reserved_mem();
 
 #ifdef CONFIG_SWIOTLB
 	swiotlb_init(1);
diff --git a/linux_5.10/arch/riscv/mm/init.c b/linux_5.10/arch/riscv/mm/init.c
index b42ccd806..d9213c839 100644
--- a/linux_5.10/arch/riscv/mm/init.c
+++ b/linux_5.10/arch/riscv/mm/init.c
@@ -195,17 +195,33 @@ void __init setup_bootmem(void)
 	setup_initrd();
 #endif /* CONFIG_BLK_DEV_INITRD */
 
-	/*
-	 * Avoid using early_init_fdt_reserve_self() since __pa() does
-	 * not work for DTB pointers that are fixmap addresses
-	 */
-	memblock_reserve(dtb_early_pa, fdt_totalsize(dtb_early_va));
-
 	if (firmware_size > PAGE_SIZE && firmware_size < LOAD_OFFSET)
 		memblock_reserve(__pa(PAGE_OFFSET), firmware_size);
 	else
 		memblock_reserve(__pa(PAGE_OFFSET), LOAD_OFFSET);
 
+	early_init_fdt_scan_reserved_mem();
+    if (!IS_ENABLED(CONFIG_BUILTIN_DTB)) {
+		/*
+		 * In case the DTB is not located in a memory region we won't
+		 * be able to locate it later on via the linear mapping and
+		 * get a segfault when accessing it via __va(dtb_early_pa).
+		 * To avoid this situation copy DTB to a memory region.
+		 * Note that memblock_phys_alloc will also reserve DTB region.
+		 */
+		if (!memblock_is_memory(dtb_early_pa)) {
+			size_t fdt_size = fdt_totalsize(dtb_early_va);
+			phys_addr_t new_dtb_early_pa = memblock_phys_alloc(fdt_size, PAGE_SIZE);
+			void *new_dtb_early_va = early_memremap(new_dtb_early_pa, fdt_size);
+
+			memcpy(new_dtb_early_va, dtb_early_va, fdt_size);
+			early_memunmap(new_dtb_early_va, fdt_size);
+			dtb_early_pa = new_dtb_early_pa;
+		} else
+			memblock_reserve(dtb_early_pa, fdt_totalsize(dtb_early_va));
+	}
+
+
 	memblock_allow_resize();
 	memblock_dump_all();
 }

```

I only need 16MB ION for my app and I was able to capture picture with the camera.
I will spend a bit more time to better understand where does this modification come from and why it is needed.

tchamelot, 2025-03-05:
Turns out i forgot some things, still trying to figure something out (sorry it was late work)