milkv-duo/duo-buildroot-sdk #176: duo-buildroot-sdk arduino suport doesnot work with arduino along with 8051 core.

jagspaul, 2025-04-02:
I have build duo-buildroot-sdk arduino suport .

Want to run both small core arduino and 8051 core.  I have gone through tutorial of DUO 8051 and modify 
/home/jagadish/embedded_linux/milk-v/duo-buildroot-sdk/build/boards/cv180x/cv1800b_milkv_duo_sd/memmap.py  as follows.

# ==============
# 8051
# ==============
CHIP8051_SIZE = 512 * SIZE_1K
CHIP8051_ADDR = DRAM_BASE + DRAM_SIZE - CHIP8051_SIZE

# ==============
# C906L FreeRTOS
# ==============
FREERTOS_SIZE = 768 * SIZE_1K
# FreeRTOS is after 8051
FREERTOS_ADDR = CHIP8051_ADDR - FREERTOS_SIZE
FSBL_C906L_START_ADDR = FREERTOS_ADDR

But it is found arduino is not working but 8051 is working.

before this modification default support was C906L FreeRTOS memmap.py was following below.

# ==============
    # C906L FreeRTOS
    # ==============
    FREERTOS_SIZE = 768 * SIZE_1K
    # FreeRTOS is at the end of DRAM
    FREERTOS_ADDR = DRAM_BASE + DRAM_SIZE - FREERTOS_SIZE
    FSBL_C906L_START_ADDR = FREERTOS_ADDR

and found arduino works perfect.
    
as per this configuration FSBL_C906L_START_ADDR = 0x83f40000
after 8051 core modification FSBL_C906L_START_ADDR = 0x83ec0000

I think arduino linker script may not support 0x83ec0000 it is still supporting 0x83f40000 instead.

However i build a baremetal ledblink with linker support 0x83ec0000 and found led bloinks works but linux ethernet IPV4 stops and showing IPV6 only. And ssh login failed for IPV6.

pls look in this matter.

Thanks & Regards
Jags

0x64620, 2025-04-04:
This is not a supported use case. You’re mixing multiple toolchains and runtime assumptions.
Fix your linker scripts. Don’t touch memmap.py unless you understand the entire boot pipeline.

jagspaul, 2025-04-04:
Thanks for your reply.
I did not do anything wrong rather I followed milk v official tutorial.

https://milkv.io/docs/duo/getting-started/8051core

In this tutorial it is clearly shown how to use custom RTOS mini core and 8051 core memory map.

However arduino is fixed with a starting memory address which is 0x83f40000 and is it not clearly mention how to modify arduion linker script to support custom starting memory address.

Thanks
Jags

0x64620, 2025-04-07:
The tutorial describes usage of 8051 core alongside minimal RTOS, not in conjunction with the Arduino runtime.

The Arduino environment is designed for the default FreeRTOS memory map. If you move the C906L base address, you must also modify:

- the Arduino linker script
- the startup code
- any hardcoded memory references in the Arduino SDK port

This is an advanced use case and not documented officially. You are combining toolchains with incompatible memory layouts.

Proceed at your own risk.