milkv-duo/duo-buildroot-sdk #179: build.sh print_err command not found

Nutz95, 2025-06-01:
I just made a git clone and tried to execute the build lunch.
It happens i get this error:
`build.sh print_err command not found`
I had to update the build.sh and rename the 4 calls from print_err to print_error
(see lines 36, 54, 70, 94.)

I was trying to build the image for emmc Arm (choice 4)

carbonfix, 2025-06-03:
This v1 sdk do not support ARM core.
Which SDK are you using now?

Nutz95, 2025-06-10:
I tried again from scratch using your documentation https://milkv.io/docs/duo/getting-started/buildroot-sdk, i followed the "Buildroot SDK V2" part.

in order to have the compilation running i had to edit the ./build.sh and edit the lines i mentioned above.
i'm running ubuntu 22.04, and i tried the choice "3. milkv-duo256m-musl-riscv64-sd"